lwfinger/rtl8723bu #94: Unreliable on Kubuntu 17.10 running 4.13 kernel

syddel, 2018-01-22:
I installed on Kubuntu 17.10 (kernel 4.13) using:

```
git clone https://github.com/lwfinger/rtl8723bu.git
cd rtl8723bu
make
sudo make install
sudo reboot
```
After rebooting, wifi routers can be seen (although each one is listed twice) and connection is successful. However, throughput is very slow, and eventually drops out altogether before coming back and then repeating (i.e. getting slow again, dropping out again).

finchMFG, 2018-04-21:
Apologies for the ping. Did you ever get this sorted?

lwfinger, 2018-04-21:
Noppe.

finchMFG, 2018-04-21:
I just got it working, I needed to blacklist my chip in `/etc/modprobe.d/blacklist.conf` like:

`
blacklist rtl8xxxu
`

Not it works like a charm on 17.10!

javlada, 2018-04-23:
For me it is still not reliable. I see two devices in Kubuntu for the single WiFi chip. It quite often disconnects by itself or the connection is just not working. Sometimes I can not connect back and only a restart of the whole tablet solves the problem.

lwfinger, 2018-04-23:
On openSUSE, it creates only one interface. The only way you would get two is if you have defined CONFIG_CONCURRENT_MODE in file include/autoconf.h. For this driver, that mode is untested.

If you have not modified  any of the code, and are still getting two interfaces, then I do not understand it.

These drivers come from Realtek, and I have no details of the programming of the wifi chip. I can help with OS crashes, but reliability issues are outside what I can fix.

javlada, 2018-04-24:
Thank you, I will check it. But i Didn't modify any code. Only disabled the default driver like finchMFG mentioned.

Apparently it is not only happening on my machine since syddel has the same problem with duplicated WiFi connections. It works most of the time but sometimes the connection gets too weak or is lost at all and I have to reboot my PC.

lwfinger, 2018-04-24:
Run the command 'dmesg > dmesg.out', put that file into some pastebin location, and post the link here.

javlada, 2018-04-29:
I finally was able to generate the debug information. It is included. If there is anything else I could do to identify the issue, please let me know.

[dmesg.out.zip](https://github.com/lwfinger/rtl8723bu/files/1959160/dmesg.out.zip)

lwfinger, 2018-05-19:
On my system - openSUSE Leap 15.0 with wireless controlled by NetworkManager - I can disconnect from an AP and connect to a different one or re-connect to that same one without any problem.

All I can conclude is that there is something wrong with the Ubuntu user-space network control software. That is beyond my control.