lwfinger/rtl8723bu #99: Cross compile arm error - ioctl_cfg80211

richgb, 2018-03-22:
Hi,

I'm using yocto to cross compile for arm and receive the following build error:

```
| make[1]: Entering directory '/home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work-shared/plnx_aarch64/kernel-build-artifacts'
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.o
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/hal/hal_phy.o
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/hal/hal_btcoex.o
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/hal/hal_usb.o
| In file included from /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/include/drv_types.h:134:0,
|                  from /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.c:22:
| /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.c: In function 'rtw_cfg80211_indicate_sta_assoc':
| /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/include/ioctl_cfg80211.h:159:60: error: implicit declaration of function 'cfg80211_send_rx_assoc' [-Werror=implicit-function-declaration]
|  #define rtw_cfg80211_send_rx_assoc(adapter, bss, buf, len) cfg80211_send_rx_assoc((adapter)->pnetdev, bss, buf, len)
|                                                             ^
| /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.c:3330:3: note: in expansion of macro 'rtw_cfg80211_send_rx_assoc'
|    rtw_cfg80211_send_rx_assoc(padapter, NULL, pmgmt_frame, frame_len);
|    ^~~~~~~~~~~~~~~~~~~~~~~~~~
| /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.c: In function 'rtw_cfg80211_indicate_sta_disassoc':
| /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.c:3391:2: error: implicit declaration of function 'cfg80211_send_disassoc' [-Werror=implicit-function-declaration]
|   cfg80211_send_disassoc(padapter->pnetdev, mgmt_buf, frame_len);
|   ^~~~~~~~~~~~~~~~~~~~~~
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/hal/hal_usb_led.o
|   CC [M]  /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/hal/HalPwrSeqCmd.o
| cc1: some warnings being treated as errors
| make[4]: *** [/home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work-shared/plnx_aarch64/kernel-source/scripts/Makefile.build:293: /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git/os_dep/ioctl_cfg80211.o] Error 1
| make[4]: *** Waiting for unfinished jobs....
| make[3]: *** [/home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work-shared/plnx_aarch64/kernel-source/Makefile:1490: _module_/home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/git] Error 2
| make[2]: *** [Makefile:150: sub-make] Error 2
| make[1]: *** [Makefile:24: __sub-make] Error 2
| make[1]: Leaving directory '/home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work-shared/plnx_aarch64/kernel-build-artifacts'
| make: *** [Makefile:338: modules] Error 2
| ERROR: Function failed: do_compile (log file is located at /home/richard/petalinux-workspace/ultrazed-wsp/build/tmp/work/plnx_aarch64-xilinx-linux/rtl8723bu/1.0-git-r0/temp/log.do_compile.9002)
```

Any ideas?

Thanks,

Richard.

richgb, 2018-03-23:
After some investigation, it appears that the undeclared functions are no longer part of the header file. Looking at past commits in this repo, I'm assuming that is what is mentioned as the new api.

I'm using a fairly recent kernel version, 4.9. So it must be something to do with the detection of the kernel version?

karora, 2018-03-23:
FWIW we're happily using these drivers with the 4.14.x series kernels, although we've patched them directly into our internal tree for the kernel build.

richgb, 2018-03-23:
Hi Karora. Thanks. I may look at doing that then, rather than compile an out of tree module.

richgb, 2018-03-23:
The issue has been resolved. I completely cleaned my project. Added it again from scratch and it works now.
Probably a dependency not built correctly or cache needed updating.

closed 2018-03-23 by richgb

honteng, 2018-12-12:
Hi,

I encountered the same issue on my cross compilation environment. Could you tell me how to resolve the issue? 

My kernel version is 4.1.15. The root cause of the issue is that cfg80211_send_rx_assoc() is removed in 4.1.15. I can find the function in version 3.x though.

I briefly looked into the driver and kernel source code, and it looks hard to retrofit the driver code to 4.1.15.

@karora Could you tell me how you figured out the issue? You say you guys applied patches and how hard it was?

@richgb Are you still using 4.9? I couldn't find the function in Linux kernel source code tree. How did you figure out the issue?