lwfinger/rtl8723bu #106: workaround: fails to compile under Fedora 28: fatal error: stdarg.h: No such file or directory

Osndok, 2018-06-04:
After updating to Fedora 28, I began receiving this compilation error:
```
make[1]: Entering directory '/usr/src/kernels/4.16.12-300.fc28.x86_64'
  CC [M]  /usr/src/rtl8723bu/core/rtw_cmd.o
In file included from ./include/linux/list.h:9,
                 from ./include/linux/rculist.h:10,
                 from ./include/linux/sched/signal.h:5,
                 from /usr/src/rtl8723bu/include/osdep_service.h:38,
                 from /usr/src/rtl8723bu/include/drv_types.h:32,
                 from /usr/src/rtl8723bu/core/rtw_cmd.c:22:
./include/linux/kernel.h:6:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:325: /usr/src/rtl8723bu/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1565: _module_/usr/src/rtl8723bu] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.16.12-300.fc28.x86_64'
make: *** [Makefile:322: modules] Error 2
```

Since there does not seem to be any reports on widescale breakage on the fedora/redhat side, I assume it has something to do with this repo.

While I do not know what the proper fix is, I find that this patch (which I hope might be useful to others) works around the issue:
```
diff --git a/Makefile b/Makefile
index e753323..d4d00b3 100644
--- a/Makefile
+++ b/Makefile
@@ -25,6 +25,7 @@ ccflags-y += -D__CHECK_ENDIAN__
 #EXTRA_CFLAGS += -Wno-uninitialized
 
 EXTRA_CFLAGS += -g -I$(src)/include
+EXTRA_CFLAGS += -I/usr/lib/gcc/x86_64-redhat-linux/8/include
 
 #EXTRA_LDFLAGS += --strip-debug
```

lwfinger, 2018-06-04:
When you see that error, do a "make clean". That should fix it.

Osndok, 2018-06-04:
I'm quite sure that I tried that several times in the course of trying to isolate this. Perhaps there is a relevant file that 'clean' is not removing?

Osndok, 2018-06-04:
I can confirm that there is a stale file in the root called ".cache.mk" that seems to be the issue... I guess it should be added to the 'make clean' rule.

lwfinger, 2018-06-04:
That file added to list to be removed when running 'make clean'. Thanks for debugging.

closed 2018-06-04 by lwfinger

AdhamRammadan, 2022-07-23:
`EXTRA_CFLAGS += -I/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.1/include/`
to the beginning of Makefile in rtl8723au directory.

lwfinger, 2022-07-24:
Adding that line to the repo's Makefile would break the build for everyone with gcc newer than 7.3.1. As I am using 12.1.1, I would expect that would be a lot of people.

AdhamRammadan, 2022-07-24:
Thanks : https://communities.vmware.com/t5/VMware-Workstation-Player/ubuntu-22-04-install-vm-workstation-erro...

This is what helped me

cd /usr/lib/vmware/modules/source
git clone https://github.com/mkubecek/vmware-host-modules
cd vmware-host-modules
git checkout workstation-16.2.3
make
tar -cf vmnet.tar vmnet-only
tar -cf vmmon.tar vmmon-only
mv vmnet.tar /usr/lib/vmware/modules/source/
mv vmmon.tar /usr/lib/vmware/modules/source/
vmware-modconfig --console --install-all