lwfinger/rtl8723bu #111: Doesn't compile for latest BBB image

gregmihalko, 2018-08-19:
Trying to compile this for the Edimax N150 2-in-1 WiFI/Bluetooth Adapter (EW-7611ULB) on a new Debian image from beagleboard.org. Clone the repo and ran `make`. Below was the output.
```
ps@beaglebone:~/rtl8723bu$ make
make: Warning: File 'Makefile' has modification time 5468759 s in the future
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.49-ti-r54/build M=/home/ps/rtl8723bu  modules
make[1]: Entering directory '/lib/modules/4.14.49-ti-r54/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.14.49-ti-r54/build'
Makefile:322: recipe for target 'modules' failed
make: *** [modules] Error 2
```
```
ps@beaglebone:~/rtl8723bu$ uname -a
Linux beaglebone 4.14.49-ti-r54 #1 SMP PREEMPT Fri Jun 15 22:14:13 UTC 2018 armv7l GNU/Linux
```
The drivers supplied by Edimax resulted in similar errors, but looking at this repo I thought there might be some more recent developments for it. Any thoughts?

gregmihalko, 2018-08-19:
Should I try to run `sudo apt-get install linux-headers-$(uname -r)` and re-try running `make` again?

lwfinger, 2018-08-19:
I'm not sure that you have the correct apt-get packages, but you do not have the kernel headers installed. That is why you are getting the error.

gregmihalko, 2018-08-19:
Thanks @lwfinger does that line above look correct? If so, I will run that and post the results.

gregmihalko, 2018-08-20:
No dice. Here's what I got:
```
ps@beaglebone:~$ sudo apt-get update
[sudo] password for ps: 
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:3 http://deb.debian.org/debian-security stretch/updates InRelease [94.3 kB]
Get:4 http://repos.rcn-ee.com/debian stretch InRelease [3,058 B]
Get:5 http://deb.debian.org/debian stretch Release [118 kB]
Get:6 http://deb.debian.org/debian stretch Release.gpg [2,434 B]
Get:7 http://deb.debian.org/debian stretch-updates/main armhf Packages [5,112 B]
Get:8 http://deb.debian.org/debian-security stretch/updates/contrib armhf Packages [1,764 B]                                                 
Get:9 http://deb.debian.org/debian-security stretch/updates/non-free armhf Packages [1,268 B]                                                
Get:10 http://deb.debian.org/debian-security stretch/updates/main armhf Packages [374 kB]                                                    
Get:11 http://repos.rcn-ee.com/debian stretch/main armhf Packages [652 kB]                                                                   
Get:12 http://deb.debian.org/debian stretch/main armhf Packages [6,927 kB]                                                                   
Get:13 http://deb.debian.org/debian stretch/non-free armhf Packages [60.0 kB]                                                                
Get:14 http://deb.debian.org/debian stretch/contrib armhf Packages [42.1 kB]                                                                 
Fetched 8,373 kB in 63d 11h 27min 21s (1 B/s)                                                                                                
Reading package lists... Done
```
```
ps@beaglebone:~$ sudo apt-get install linux-headers-$(uname -r)
[sudo] password for ps: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  linux-headers-4.14.49-ti-r54
0 upgraded, 1 newly installed, 0 to remove and 39 not upgraded.
Need to get 11.2 MB of archives.
After this operation, 77.6 MB of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf linux-headers-4.14.49-ti-r54 armhf 1stretch [11.2 MB]
Fetched 11.2 MB in 4s (2,437 kB/s)                       
Selecting previously unselected package linux-headers-4.14.49-ti-r54.
(Reading database ... 34916 files and directories currently installed.)
Preparing to unpack .../linux-headers-4.14.49-ti-r54_1stretch_armhf.deb ...
Unpacking linux-headers-4.14.49-ti-r54 (1stretch) ...
Setting up linux-headers-4.14.49-ti-r54 (1stretch) ...
```
Which seemed like a good result, but:
```
ps@beaglebone:~/rtl8723bu$ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.49-ti-r54/build M=/home/ps/rtl8723bu  modules
make[1]: Entering directory '/lib/modules/4.14.49-ti-r54/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.14.49-ti-r54/build'
Makefile:322: recipe for target 'modules' failed
make: *** [modules] Error 2
```

lwfinger, 2018-08-20:
You need to find out the steps in preparing builds for out-of-kernel modules from your distro. If you were using openSUSE, I could help, but not for any .deb distro.

gregmihalko, 2018-08-20:
Got it - where would I start to find that info? Would it be something that could be incorporated back into this repo? If we ended up with good documentation, this adapter would be an awesome addition to standard BBB iot stacks!

lwfinger, 2018-08-20:
Search for "building external modules" or "kernel headers" for your distro. Alternatively, you might check the forums of your distro. There should not be a difference between X86 and ARM.