lwfinger/rtl8723bu #114: Only works after suspending my computer

owazae, 2018-09-15:
Hello,
Here is my problem : After launching KDE, wifi is trying to connect but unsuccessfully.
It will correctly connect after I suspend the computer, then enter KDE again.

I'm kind of beginner on linux, so I have no idea where that problem comes from. Could someone help me ?

> 4.15.0-34-generic #37-Ubuntu SMP Mon Aug 27 15:21:48 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

lwfinger, 2018-09-15:
Your report is very sketchy, and short on details.

What driver are you using? As you posted here, I would assume that you are trying to use this driver. In that case, what have you done to prevent the other driver (rtl8xxxu) from loading? If you are really using rtl8xxxu, then post your complaint at linux-wireless@vger.kernel.org.

Unsure about the driver - post the outpuf of the command 'lsmod | egrep "8723|8xxx".

owazae, 2018-09-15:
I'm using this driver, I followed the steps for manual installation.

This is the result of the command : 

> rtl8xxxu              122880  0
> mac80211              778240  1 rtl8xxxu

I'm not sure how to give more details, I find wifi networks around me by but I'm unable to connect to any if I haven't suspended the laptop before. So each time I reboot, I have to put the computer in stand by  before connecting to wifi network.
It is really weird, I may have done something wrong during the installation, but I don't know what.

karora, 2018-09-15:
Your problem is that you *aren't* using this driver.  You need to blacklist the rtl8xxxu module and load the 8723bu module instead.

lwfinger, 2018-09-16:
Well said.

owazae, 2018-09-16:
Ok thank you guys, I'll send an email

closed 2018-09-22 by owazae