lwfinger/rtl8723bu #116: no scan results

sunskyhsh, 2018-10-15:
I have installed the driver on a tablet EZpad 6 pro with Ubuntu 18.04. But it shows no scan results when I type sudo iwlist scan.

lwfinger, 2018-10-15:
Are you sure you have an rtl8723bu? Post the output of the command 'lsusb'.

sunskyhsh, 2018-10-15:
```
bit@bit-EZpad:~$ lsusb
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 006: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 001 Device 004: ID 058f:5608 Alcor Micro Corp. 
Bus 001 Device 009: ID 0bda:8152 Realtek Semiconductor Corp. 
Bus 001 Device 007: ID 1c4f:0026 SiGma Micro Keyboard
Bus 001 Device 005: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0781:5590 SanDisk Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

lwfinger, 2018-10-15:
The 0bda:b720 device is an RTL8723BU. Now post the output of the dmesg command in some pastebin location and post the link.

sunskyhsh, 2018-10-15:
https://paste.ubuntu.com/p/cVHFH4QKBj/

lwfinger, 2018-10-15:
Your dmesg buffer must be quite small. It has wrapped around and is full of sequences like
[   21.056694] ====>_WriteFW 201
[   21.057059] ====>_BlockWrite 95 i:1

Those did not come from this driver. In fact, I see no output from this driver. For completeness, please post the output of the 'lsmod' command so that I can see which drivers are loaded.

sunskyhsh, 2018-10-15:
This is my lsmod & dmesg output:
https://paste.ubuntu.com/p/vCyDB8pYPh/

lwfinger, 2018-10-15:
I see that 8723bu is loaded, but once again your dmesg shows nothing useful. You clearly are presenting only part of the dmesg output. That is OK, but I will be unable to help you.

odoral, 2019-03-15:
Same issue here:

`pi@jarvis:~ $ lsusb
Bus 001 Device 004: ID 0bda:b720 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. SMC9512/9514 USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

`[   14.528061] Bluetooth: HCI device and connection manager initialized
[   14.528096] Bluetooth: HCI socket layer initialized
[   14.528117] Bluetooth: L2CAP socket layer initialized
[   14.528174] Bluetooth: SCO socket layer initialized
[   15.379338] usbcore: registered new interface driver btusb
[   15.388658] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[   15.388683] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[   15.402324] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   15.402363] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   15.407120] Bluetooth: hci0: rom_version status=0 version=1
[   15.407236] Bluetooth: cfg_sz 0, total size 22496
[   16.340403] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   17.859279] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[   20.078734] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   20.815354] 8723bu: loading out-of-tree module taints kernel.
[   21.944885] RTL871X: module init start
[   21.944908] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   21.944916] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   23.026682] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.026701] Bluetooth: BNEP filters: protocol multicast
[   23.026731] Bluetooth: BNEP socket layer initialized
[   24.178043] RTL871X: rtw_ndev_init(wlan0)
[   24.283630] RTL871X: rtw_ndev_init(wlan1)
[   24.356156] usbcore: registered new interface driver rtl8723bu
[   24.356176] RTL871X: module init ret=0
[   27.903663] RTL871X: RTW_ADAPTIVITY_EN_
[   27.903691] AUTO, chplan:0x20, Regulation:3,3
[   27.903701] RTL871X: RTW_ADAPTIVITY_MODE_
[   27.903706] NORMAL
[   28.593067] random: crng init done
[   28.593093] random: 7 urandom warning(s) missed due to ratelimiting
[   29.293304] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   30.504283] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   32.564949] RTL871X: nolinked power save enter`

odoral, 2019-03-15:
I've just seen your comment in https://github.com/lwfinger/rtl8723bu/issues/46#issuecomment-255596603 and checked that `sudo iw dev wlan0 scan` is working.

lwfinger, 2019-03-15:
If the scan using iw is working, then the driver is working! It seems that the problem is with your distro;s network management packages. Sorry, but I cannot do anything with those. I use openSUSE with NetworkManager, and the driver/management software all work.

odoral, 2019-03-16:
The problem in my case was fixed using sudo. Thank you for your support!!!!