lwfinger/rtl8723bu #117: Build fails in Xubuntu 18.04, kernel v4.19.0-041900-generic

agats, 2018-10-26:
First time trying to build this, but I am getting the following error when running `make`.

```
/home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.c:699:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
```

The full log is below

```
drew@drew-ALPHA:~/Downloads/Litebook/rtl8723bu$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-041900-generic/build M=/home/drew/Downloads/Litebook/rtl8723bu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-041900-generic'
  CC [M]  /home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.o
In file included from /home/drew/Downloads/Litebook/rtl8723bu/include/drv_types.h:35:0,
                 from /home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.c:22:
/home/drew/Downloads/Litebook/rtl8723bu/include/wifi.h:831:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from ./include/net/cfg80211.h:24:0,
                 from /home/drew/Downloads/Litebook/rtl8723bu/include/osdep_service_linux.h:74,
                 from /home/drew/Downloads/Litebook/rtl8723bu/include/osdep_service.h:40,
                 from /home/drew/Downloads/Litebook/rtl8723bu/include/drv_types.h:32,
                 from /home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.c:22:
./include/linux/ieee80211.h:1442:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.c:699:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.c:699:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:305: recipe for target '/home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.o' failed
make[2]: *** [/home/drew/Downloads/Litebook/rtl8723bu/os_dep/os_intfs.o] Error 1
Makefile:1517: recipe for target '_module_/home/drew/Downloads/Litebook/rtl8723bu' failed
make[1]: *** [_module_/home/drew/Downloads/Litebook/rtl8723bu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-041900-generic'
Makefile:322: recipe for target 'modules' failed
make: *** [modules] Error 2
```

agats, 2018-10-26:
I was able to get the project to build after making these changes.

```
diff --git a/os_dep/os_intfs.c b/os_dep/os_intfs.c
index 44cd53b..750ffba 100644
--- a/os_dep/os_intfs.c
+++ b/os_dep/os_intfs.c
@@ -695,9 +695,6 @@ static const struct net_device_ops rtw_netdev_ops = {
        .ndo_open = netdev_open,
        .ndo_stop = netdev_close,
        .ndo_start_xmit = rtw_xmit_entry,
-#if (LINUX_VERSION_CODE>=KERNEL_VERSION(2,6,35))
-       .ndo_select_queue       = rtw_select_queue,
-#endif
        .ndo_set_mac_address = rtw_net_set_mac_address,
        .ndo_get_stats = rtw_net_get_stats,
        .ndo_do_ioctl = rtw_ioctl,
```

lwfinger, 2018-10-26:
It is generally not advised to remove features just to get the code to build.

I just pushed a better fix. :)

closed 2018-10-26 by lwfinger

agats, 2018-10-26:
Thanks @lwfinger, I agree with
> It is generally not advised to remove features just to get the code to build.
I just really needed my laptop to work. And thank you for the quick and proper fix!