lwfinger/rtl8723bu #118: Wifi interface twice for kernel 4.4.0-138.164

Adrian-Bielefeldt, 2018-10-26:
LInux Mint 18.1 XFCE

When upgrading from 4.4.0-53.74 to 4.4.0-138.164, I noticed that it shows the Wifi interface twice. I can generally connect, but get timeouts often. If I disconnect and then reconnect, networking works again for a short time. Reverting to 4.4.0-53.74 removed the second interface again.

lwfinger, 2018-10-26:
I have no idea what Mint changed between 4.4.0-53 and 4.4.0-138. Does Mint provide a git repo containing all their changes so that you could bisect to find the exact change that introduces the error?

Adrian-Bielefeldt, 2018-10-26:
I'm using Linux Mint (Ubuntu Edition), so that should be the normal Ubuntu [repo](https://launchpad.net/ubuntu/+source/linux) on launchpad. The error appears when upgrading from [4.4.0-53.74](https://launchpad.net/ubuntu/+source/linux/4.4.0-53.74) to [4.4.0-138.164](https://launchpad.net/ubuntu/+source/linux/4.4.0-138.164).

I am not familiar with launchpad, but it should be possible to find all changes occuring from 53.73 to 138.164. If I can help out with any logs or traces I will gladly do so.

lwfinger, 2018-10-27:
As you are the one with the problem, it will be up to you to find the cause.

javlada, 2018-10-29:
@Adrian-Bielefeldt I have the same problem as you on my Kubuntu. There were other people reporting the same. It seems to be a general *ubuntu problem and unfortunately there is no solution AFAIK.

Adrian-Bielefeldt, 2018-10-31:
While trying to learn how to increment my kernel version, a user in this [thread](https://forums.linuxmint.com/viewtopic.php?f=46&t=280425&p=1546683#p1544781) on [forums.linuxmint.org](forums.linuxmint.org) proposed to comment out this line in the Makefile, so change it from
`EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE`
to
`#EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE`
and rebuild the driver. This worked for me.

I do not know what this option does or if I loose any features disabling it, but at least my wifi is stable (and shows only one interface).

lwfinger, 2018-10-31:
That option creates two "concurrent" interfaces. Usually one is used for a station and the second for an AP. If you have trouble with stability, then disabling that feature is reasonable.

closed 2018-11-07 by Adrian-Bielefeldt