lwfinger/rtl8723bu #120: Unable to start AP mode on OpenWrt

terrycmchan, 2018-11-06:
Hi lwfinger,

I ported this driver on OpenWrt (Kernel version: 4.1.0).
However, the AP mode could not be started. I got the below error:
>  kern.warn kernel: [   31.970000] RTL871X: cfg80211_rtw_add_virtual_intf(wlan0) wiphy:phy0, name:wlan0, type:3
> kern.warn kernel: [   31.980000] RTL871X: cfg80211_rtw_add_virtual_intf(wlan0) ndev:  (null), ret:-19
> daemon.notice netifd: radio1 (1235): command failed: No such device (-19)

I found that the AP mode seems not to be supported in the function "cfg80211_rtw_add_virtual_intf()" in the file "ioctl_cfg80211.c"
> 	case NL80211_IFTYPE_AP:
> 		ret = -ENODEV;
> 		break;

What have I missed since the AP mode seems to be supported mentioned on other threads?

Thanks!

lwfinger, 2018-11-06:
I'm not sure what is/was missed. In looking at v5.2.2.4 of the driver for a different device, this routine is considerably changed. I do not have the time now to analyze the differences, but I will look at it in the next couple of days. More later.

terrycmchan, 2018-11-07:
Thanks.
And, may I know where I can find the v5.2.2.4?

superwinnieperruna, 2018-11-18:
Hello lwfinger,i am on the same condition as terrycmchan,i am trying use AP mode for RT8723BU but it can't be started, ,,, how to ?  ... thanks in advance ,cheers

lwfinger, 2018-11-19:
Are you using hostapd? What driver? Which branch?

terrycmchan, 2018-11-19:
Hi lwfinger,
The hostapd version is openwrt's hostapd-2015-03-25.
The driver is the master branch of lwfinger/rtl8723bu.

lwfinger, 2018-11-19:
The master branch is an older driver. You should use the v4.0.10 branch. I know of at least one vendor that uses that branch with openWRT to create a VPN appliance.

superwinnieperruna, 2018-11-19:
4.3 branch would work too?

lwfinger, 2018-11-19:
In earlier testing, 4.3 had some problems that I have not had time to fix. Please use v4.0.10 as I recommended.

superwinnieperruna, 2018-11-19:
Excuse me dumb question ,where to download It ?

lwfinger, 2018-11-19:
Please read the git manual, but here is the command:

git clone https://github.com/lwfinger/rtl8723bu.git -b v4.0.10

If you already have used git to clone the driver, then

git checkout origin/v4.0.10 -b v4.0.10

terrycmchan, 2018-11-20:
Hi lwfinger,

I got the below error:
Cloning into 'rtl8723bu'...
fatal: Remote branch v4.0.10 not found in upstream origin

lwfinger, 2018-11-20:
I got confused. What you should use for rtl8723bu is v4.3.16.

superwinnieperruna, 2018-11-20:
i will try to check with this version if AP mode works, did you include Bluetooth function on this driver?

lwfinger, 2018-11-21:
I did not include anything. All the drivers come from Realtek. My only contribution is to make them available and to be sure that they are modified to account for changing kernel APIs, and to try to fix any oops reports that are posted. With respect to functional problems like AP mode failing, I lack the knowledge of the device and the driver to fix those.

githaff, 2019-01-23:
Hi terrycmchan. I also work on the rtl8723bu with the OpenWRT (mine is the 18.06.1). What did you do to port it? I currently see errors like
```
8723bu: disagrees about version of symbol cfg80211_scan_done
8723bu: Unknown symbol cfg80211_scan_done (err -22)
```
when I load the driver. As I understand that is because of disagreements with the mac80211 backports.

EdKee, 2019-07-11:
githaff, you must be compiling with wrong kernel sources or kernel config.

EdKee, 2019-07-11:
Hi,

is anyone able to use this card with Ubuntu 18.4 in AP/Hotspot mode with WPA2? 
Actually I need both client and AP mode from ARM machine, but not simultaneously. Either connection to infrastructure or tethering ethernet connection. And there are problems with tethering. Ubuntu on PC network manager can be persuaded to disable cipher, after which at least Windows 10 client PC can connect and work. Android 8 phone seems constantly dropping connection. Things change if WPA2 is enabled on rtl8723bu AP. Then both Android 8 and Windows PC keep dropping connection. 
Is there any way to fix driver problem?

When cipher turned off using Ubuntu nm-connection-editor command, card plug-in and Ubuntu Hotpot enable produces this. It is LM811 module, please disregard lines about Bluetooth:

[11397.524047] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[11397.790588] usb 1-2: New USB device found, idVendor=0bda, idProduct=b720, bcdDevice= 2.00
[11397.790600] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11397.790609] usb 1-2: Product: 802.11n WLAN Adapter
[11397.790618] usb 1-2: Manufacturer: Realtek
[11397.790627] usb 1-2: SerialNumber: 00e04c000001
[11397.806441] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[11397.806449] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[11397.806501] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[11397.806509] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[11397.808747] Bluetooth: hci0: rom_version status=0 version=1
[11397.808765] Bluetooth: hci0: cfg_sz 0, total size 22496
[11398.121788] RTL871X: rtw_ndev_init(wlan0)
[11398.167045] usbcore: registered new interface driver rtl8xxxu
[11398.186021] rtl8723bu 1-2:1.2 wlx34c9f08a5948: renamed from wlan0
[11398.226522] IPv6: ADDRCONF(NETDEV_UP): wlx34c9f08a5948: link is not ready
[11399.082980] RTL871X: RTW_ADAPTIVITY_EN_
[11399.082983] AUTO, chplan:0x20, Regulation:0,0
[11399.082984] RTL871X: RTW_ADAPTIVITY_MODE_
[11399.082984] NORMAL
[11399.886528] IPv6: ADDRCONF(NETDEV_UP): wlx34c9f08a5948: link is not ready
[11399.943861] IPv6: ADDRCONF(NETDEV_UP): wlx34c9f08a5948: link is not ready
[11403.289499] RTL871X: nolinked power save enter
[11415.407453] RTL871X: RTW_ADAPTIVITY_EN_
[11415.407457] AUTO, chplan:0x20, Regulation:0,0
[11415.407457] RTL871X: RTW_ADAPTIVITY_MODE_
[11415.407458] NORMAL
[11416.106708] RTL871X: nolinked power save leave
[11416.345892] netlink: 'wpa_supplicant': attribute type 213 has an invalid length.
[11416.611742] RTL871X: assoc success
[11416.612106] IPv6: ADDRCONF(NETDEV_CHANGE): wlx34c9f08a5948: link becomes ready

How to fix "netlink: 'wpa_supplicant': attribute type 213 has an invalid length."?

As I said with cipher off Windows client seems working well. Android 9 phone keeps dropping connection without any sign in dmesg. 

Now enabling Hotspot with WPA enabled. 

[12257.890296] netlink: 'wpa_supplicant': attribute type 213 has an invalid length.
[12258.161792] RTL871X: assoc success
[12258.162231] IPv6: ADDRCONF(NETDEV_CHANGE): wlx34c9f08a5948: link becomes ready
[12258.372133] RTL871X: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:TKIP
[12304.527109] RTL871X: send eapol packet
[12304.531438] RTL871X: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:TKIP
[12304.625379] RTL871X: send eapol packet
[12304.632538] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12311.431222] RTL871X: send eapol packet
[12311.538480] RTL871X: send eapol packet
[12311.544937] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12318.454755] RTL871X: send eapol packet
[12318.506399] RTL871X: send eapol packet
[12318.513613] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12328.266951] RTL871X: send eapol packet
[12328.390033] RTL871X: send eapol packet
[12328.496303] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12335.160348] RTL871X: send eapol packet
[12336.171068] RTL871X: send eapol packet
[12336.182419] RTL871X: send eapol packet
[12336.188697] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12345.957006] RTL871X: send eapol packet
[12346.068733] RTL871X: send eapol packet
[12346.082947] RTL871X: set pairwise key camid:4, addr:f4:60:e2:00:00:00, kid:0, type:AES
[12352.893982] RTL871X: send eapol packet
[12353.389322] RTL871X: ap recv deauth reason code(1) sta:f4:60:e2:00:00:00
[12353.389567] RTL871X: clear key for addr:f4:60:e2:f0:29:10, camid:4

Why first two group key lines are TKIP? Is it because of netlink attribute type 213 error?
Those 6 with AES are reconnection attempts. Deauth comes when Android decides to choose working network.

Any ideas?

EdKee, 2019-07-12:
I'll start new issue, since "set pairwise key camid" and "send eapol packet" seem not being the problem. Android 4 phone as a client to RTL8723BU equipped Ubuntu hotspot works +- well. Windows 10 PC client after few connection attempts also seems working well. Andoid 8 phone drops connection and switches to different AP or cellular instead. 
What's interesting in case of Android 4 groups of eapol-like messages come about each half of minute. Windows and Android 8 about each 10 seconds. Perhaps Android 8 doesn't like absence of WPA2 support. iw list command gives this in ciphers section:

	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)

Is something missing here for WPA2 support on running card in AP mode?