lwfinger/rtl8723bu #121: cpu stall when enabled CONFIG_USE_USB_BUFFER_ALLOC_RX

smzahraee, 2018-12-10:
in driver initilization if i enabled `CONFIG_USE_USB_BUFFER_ALLOC_RX` the function `rtw_os_recvbuf_resource_alloc` will return `_FAIL` and finally we get **cpu stall** error message in kernel.

smzahraee, 2018-12-11:
i use this driver in sunxi a20 with kernel 3.4.
i disabled CONFIG_USE_USB_BUFFER_ALLOC_RX and every thigs is fine now . but i don't know its side effects!!!

lwfinger, 2018-12-11:
Once there is a CPU stall, then unloading the driver is unlikely to correct the stall. From what I see in the code, if the driver is unable to allocate DMA coherent memory, the driver does not make a proper recovery. Rather than trying to fix that situation, I will remove all usage of CONFIG_USE_USB_BUFFER_ALLOC_RX so that no one else falls into this trap. I do not think there are serious side effects. DMA with USB devices should work without needing coherent memory.

closed 2018-12-11 by lwfinger