lwfinger/rtl8723bu #123: Unable to compile drivers on linux kernel 4.19 and higher

bonnom, 2018-12-15:
First of all thank you for making this driver I have however a problem with building the driver for linux kernel 4.19.x. I have no problems with building the driver for the linux kernel 4.18 and below.

I only have this problem with this driver, other drivers build just fine.

Below I pasted the beginning and the end of the console, hopefully it provides some useful information!

Operating system:
KDE neon (bionic)

```
user@Computer:~/rtl8723bu$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.9-041909-generic/build M=/home/user/rtl8723bu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.9-041909-generic'
  CC [M]  /home/user/rtl8723bu/core/rtw_cmd.o
In file included from /home/user/rtl8723bu/include/drv_types.h:35:0,
                 from /home/user/rtl8723bu/core/rtw_cmd.c:22:
/home/user/rtl8723bu/include/wifi.h:831:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from ./include/net/cfg80211.h:24:0,
                 from /home/user/rtl8723bu/include/osdep_service_linux.h:74,
                 from /home/user/rtl8723bu/include/osdep_service.h:40,
                 from /home/user/rtl8723bu/include/drv_types.h:32,
                 from /home/user/rtl8723bu/core/rtw_cmd.c:22:
./include/linux/ieee80211.h:1442:0: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /home/user/rtl8723bu/core/rtw_security.o
In file included from /home/user/rtl8723bu/include/drv_types.h:35:0,
                 from /home/user/rtl8723bu/core/rtw_security.c:22:
/home/user/rtl8723bu/include/wifi.h:831:0: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

…

/home/user/rtl8723bu/os_dep/os_intfs.c:699:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/home/user/rtl8723bu/os_dep/os_intfs.c:699:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/user/rtl8723bu/os_dep/os_intfs.o' failed
make[2]: *** [/home/user/rtl8723bu/os_dep/os_intfs.o] Error 1
Makefile:1517: recipe for target '_module_/home/user/rtl8723bu' failed
make[1]: *** [_module_/home/user/rtl8723bu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.9-041909-generic'
Makefile:322: recipe for target 'modules' failed
make: *** [modules] Error 2b

```

bonnom, 2018-12-15:
Sorry seems you already fixed it!!

lwfinger, 2018-12-15:
Yes I have. You need to use git to clone the source, and ALWAYS di a 'git pull' before you report any "error" like this.

closed 2018-12-15 by lwfinger