lwfinger/rtl8723bu #124: Rebuilding is now failing

fhdk, 2018-12-19:
As a Manjaro user I rebuild the module on a regular basis.

After the recent change to the source I am getting a bunch of errors like this

```
  CC [M]  /build/linux419-rtl8723bu/src/rtl8723bu/hal/odm_RegConfig8723B.o
objdump: /build/linux419-rtl8723bu/src/rtl8723bu/hal/.tmp_HalBtc8723b2Ant.o: unable to initialize decompress status for section .debug_info
objdump: /build/linux419-rtl8723bu/src/rtl8723bu/hal/.tmp_HalBtc8723b2Ant.o: unable to initialize decompress status for section .debug_info
objdump: /build/linux419-rtl8723bu/src/rtl8723bu/hal/.tmp_HalBtc8723b2Ant.o: file format not recognized
```

lwfinger, 2018-12-19:
You have a corrupted set of objects. The following should help:

git pull
sudo make clean
make
sudo make install

fhdk, 2018-12-19:
OK. I can understand that - but the issue comes from downloading the master.zip - extracting and building - so where does the issue come from. Does something go wrong when pulling the zip?

lwfinger, 2018-12-19:
Why a chroot environment. While logged in as a normal user, just do "make" followed by "sudo make install". The build part should never be run as root, only the install.

I have no idea what is happening. Does it still happen when you do not use chroot? As far as I know, a normal build should not use objdump.

fhdk, 2018-12-19:
I found that it is an issue with Arch Linux.

https://bbs.archlinux.org/viewtopic.php?id=242682

closed 2018-12-19 by fhdk