lwfinger/rtl8723bu #125: compile error: objdump: ... : file format not recognized

cypa, 2018-12-21:
[make.log](https://github.com/lwfinger/rtl8723bu/files/2703287/make.log)
strange error appeared after arch linux update to 4.19.11-arch1-1-ARCH kernel, see make.log attached (in russian)
error translates to 

objdump: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/.tmp_rtw_security.o: unable to initialize decompress status for section .debug_info
objdump: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/.tmp_rtw_security.o: unable to initialize decompress status for section .debug_info
objdump: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/.tmp_rtw_security.o: file format not recognized

and so on

cypa, 2018-12-21:
installed elfutils but th error persists

hyphenc, 2018-12-21:
I encountered the same issue when upgrading to 4.19.11-arch1-1-ARCH. Downgrading both elfutils and libelf to 0.174-1 (+ adding them to IgnorePkg) and then upgrading linux, linux-headers, etc. worked for me. After rebooting, the driver compiled just fine.

cypa, 2018-12-22:
yes, that works, but what should we do next?

lwfinger, 2018-12-22:
Complain to ARCH. They are the ones that screwed something up.

pseudotheist, 2019-08-20:
So, I had a similar issue trying to install this drive in KaOS:
[Install script](https://pastebin.com/FHmdjGNq)
But I'm not sure how it would be an elfutils issue, since the driver installed fine on Ubuntu Disco Dingo, and it seems they both have elfutils version 176-1.

lwfinger, 2019-08-21:
This is more likely to be a compiler problem. Do they have the same version of gcc?

pseudotheist, 2019-08-22:
So far as I can tell, they seem to be the same version:
Ubuntu: "Version: 4:8.3.0-1ubuntu3"
KaOS: 8.3.0-1