lwfinger/rtl8723bu #126: Cannoct connect to AP on kernel 4.19

sgufa, 2019-01-07:
On kernel 4.19 when i connect using wps_pbc i get "<3>Association request to the driver failed" and card cannot connect to the Access point. Tried with various APs. If i create network manually and set ssid and psk it works ok.
Any help?

lwfinger, 2019-01-07:
Obviously, there is a bug in the WPS implementation. I doubt it gets much checking as the use of WPS is considered a big security hole. Nearly all of my testing is done using NetworkManager, which does most of the work in creating the connection. All I need to do is key in the secret.

A quick search fails to find any similar reports. I will take a quick look at the code, but I make no promises.

sgufa, 2019-01-07:
fyi i get
<3>CTRL-EVENT-EAP-STATUS status='completion' parameter='failure'
<3>CTRL-EVENT-EAP-FAILURE EAP authentication failed
and then the association request failure message.

sgufa, 2019-01-07:
fyi 2: tried rtl8xxxu generic driver in the kernel and it works with WPS

sgufa, 2019-01-08:
4.3.16 doesnt compile on 4.19, i get:
"rtl8723bu/os_dep/ioctl_cfg80211.c:6354:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
/rtl8723bu/os_dep/ioctl_cfg80211.c:6354:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6377:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .del_virtual_intf = cfg80211_rtw_del_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6377:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors"

sgufa, 2019-01-08:
i used EXTRA_CFLAGS += -Wno-error=incompatible-pointer-types in Makefile but now i get other errors:

`/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6354:25: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]

                         ^~~~~~~~~~
/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6354:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6377:22: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .del_virtual_intf = cfg80211_rtw_del_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/lw/rtl8723bu/os_dep/ioctl_cfg80211.c:6377:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
  CC [M]  /root/lw/rtl8723bu/os_dep/rtw_android.o
  CC [M]  /root/lw/rtl8723bu/os_dep/rtw_proc.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_intf.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_com.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_com_phycfg.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_phy.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_btcoex.o
  CC [M]  /root/lw/rtl8723bu/hal/hal_usb.o
/root/lw/rtl8723bu/hal/hal_btcoex.c:124:18: error: ‘COMP_MAX’ undeclared here (not in a function)
 u32 GLBtcDbgType[COMP_MAX];
                  ^~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:125:16: error: ‘BT_TMP_BUF_SIZE’ undeclared here (not in a function)
 u8 GLBtcDbgBuf[BT_TMP_BUF_SIZE];
                ^~~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c: In function ‘halbtcoutsrc_AggregationCheck’:
/root/lw/rtl8723bu/hal/hal_btcoex.c:370:21: error: ‘BTC_BT_INFO {aka struct _BTC_BT_INFO}’ has no member named ‘bPreRejectAggPkt’; did you mean ‘bRejectAggPkt’?
   pBtCoexist->btInfo.bPreRejectAggPkt = pBtCoexist->btInfo.bRejectAggPkt;
                     ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:374:24: error: ‘BTC_BT_INFO {aka struct _BTC_BT_INFO}’ has no member named ‘bPreRejectAggPkt’; did you mean ‘bRejectAggPkt’?
   if(pBtCoexist->btInfo.bPreRejectAggPkt)
                        ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:377:22: error: ‘BTC_BT_INFO {aka struct _BTC_BT_INFO}’ has no member named ‘bPreRejectAggPkt’; did you mean ‘bRejectAggPkt’?
    pBtCoexist->btInfo.bPreRejectAggPkt = pBtCoexist->btInfo.bRejectAggPkt;
                      ^
/root/lw/rtl8723bu/hal/hal_btcoex.c: In function ‘halbtcoutsrc_Get’:
/root/lw/rtl8723bu/hal/hal_btcoex.c:716:8: error: ‘BTC_GET_BL_IS_ASUS_8723B’ undeclared (first use in this function)
   case BTC_GET_BL_IS_ASUS_8723B:
        ^~~~~~~~~~~~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:716:8: note: each undeclared identifier is reported only once for each function it appears in
/root/lw/rtl8723bu/hal/hal_btcoex.c:789:8: error: ‘BTC_GET_U1_ANT_TYPE’ undeclared (first use in this function)
   case BTC_GET_U1_ANT_TYPE:
        ^~~~~~~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:793:24: error: ‘BTC_ANT_TYPE_0’ undeclared (first use in this function)
      *pU1Tmp = (u1Byte)BTC_ANT_TYPE_0;
                        ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:794:27: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
      pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_0;
                           ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:797:24: error: ‘BTC_ANT_TYPE_1’ undeclared (first use in this function)
      *pU1Tmp = (u1Byte)BTC_ANT_TYPE_1;
                        ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:798:27: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
      pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_1;
                           ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:801:24: error: ‘BTC_ANT_TYPE_2’ undeclared (first use in this function)
      *pU1Tmp = (u1Byte)BTC_ANT_TYPE_2;
                        ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:802:27: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
      pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_2;
                           ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:805:24: error: ‘BTC_ANT_TYPE_3’ undeclared (first use in this function)
      *pU1Tmp = (u1Byte)BTC_ANT_TYPE_3;
                        ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:806:27: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
      pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_3;
                           ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:809:24: error: ‘BTC_ANT_TYPE_4’ undeclared (first use in this function)
      *pU1Tmp = (u1Byte)BTC_ANT_TYPE_4;
                        ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:810:27: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
[... 472 lines cut ...]
/root/lw/rtl8723bu/hal/hal_btcoex.c:3176:26: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
     pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_3;
                          ^
/root/lw/rtl8723bu/hal/hal_btcoex.c:3176:45: error: ‘BTC_ANT_TYPE_3’ undeclared (first use in this function)
     pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_3;
                                             ^~~~~~~~~~~~~~
/root/lw/rtl8723bu/hal/hal_btcoex.c:3179:26: error: ‘BTC_BOARD_INFO {aka struct _BTC_BOARD_INFO}’ has no member named ‘antType’
     pBtCoexist->boardInfo.antType = (u1Byte)BTC_ANT_TYPE_4;
                          ^
[... cut ...]
/root/lw/rtl8723bu/hal/hal_usb_led.c:5156:2: error: implicit declaration of function ‘_init_timer’ [-Werror=implicit-function-declaration]
  _init_timer(&(pLed->BlinkTimer), padapter->pnetdev, BlinkTimerCallback, pLed);
  ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/root/lw/rtl8723bu/hal/hal_usb_led.o' failed
make[2]: *** [/root/lw/rtl8723bu/hal/hal_usb_led.o] Error 1
Makefile:1517: recipe for target '_module_/root/lw/rtl8723bu' failed
make[1]: *** [_module_/root/lw/rtl8723bu] Error 2`

sgufa, 2019-01-10:
Hello. tried [https://github.com/ulli-kroll/rtl8723bu](url) and it works good. also WPS is working.

lwfinger, 2019-01-10:
Use whatever works.

closed 2019-01-10 by lwfinger