lwfinger/rtl8723bu #132: bad RX (download) performance

maurerr, 2019-03-01:
on my ubuntu 18.04 with kernel 4.15-45 the best results i've got is 14mbps DL and 40mbps UL.
The UL speed is the same as in Windows 10 but the RX/download is quite bad.
Is there something we could try to improve this performance?

lwfinger, 2019-03-01:
I have no idea why the performance is not what you expect. However, I am not privy to any of the internals of the chip, and I did not write the driver. My only involvement is to ensure that the driver compiles on all versions of the kernel, and that it does not crash.

I will measure my throughput with the driver and let you know what I get.

lwfinger, 2019-03-02:
Ok, I measure about 24 Mbps DL. The upload is the full speed of my broadband connection, and the download is actually faster than I get on my iPhone.

With a late-version kernel, you need to be careful that you are not having driver rtl8xxxu loaded. That one is known to have low performance. Note that I have nothing to do with that driver.

My AP is a Netgear WNDR4500, which is an 802.11n router. Getting an improvement in the performance of your connection would require a comparison of the network traffic between the Windows driver and that of the Linux driver using Wireshark to capture the over-the-air packets. Not only is it difficult to capture those packets, but the analysis would be very demanding of time. Unless, you have a lot of time to do this yourself, or a lot of money to contract out the effort, I am afraid you will need to live with the performance.

maurerr, 2019-03-02:
A big thank you for your comment and all your work.
If this driver didn't existed rtl8xxxu would crash all the time for me.
What distro and kernel are you using to get 24mbps?

lwfinger, 2019-03-02:
I am running openSUSE Tumbleweed using the v5.0.0-rc8 kernel. The result was gotten with the web site speedtest.net. Note that the speed should not be highly dependent on the kernel or distro since the main results come from the driver. It could depend on the CPU speed. According to /proc/cpuinfo, mine is "Intel(R) Core(TM) i7-4600M CPU @ 2.90GHz", which is reasonably fast.

maurerr, 2019-03-03:
indeed i have low-end n3350 so that might explain the low throughput.
In the meantime I've checked again in a free environment (no other APs around) and got almost 19mbps.
Thanks for all your input

closed 2019-03-03 by maurerr