lwfinger/rtl8723bu #134: Is the driver contains RTL8723BU Bluetooth part？

wwfdzh2015, 2019-03-20:
hello
 my device is RTL8723BU， it contains wifi and Bluetooth.I want to run it on embedded linux4.6 kernel.Dose the driver support it?
thank you!

lwfinger, 2019-03-20:
This driver supports the RTL8723BU wifi part. The Bluetooth part needs a different driver. I used to provide a driver for the BT part, but that is now included as part of the btusb driver in later kernels. You will need to backport that code into the 4.6 kernel.

closed 2019-03-20 by lwfinger

wwfdzh2015, 2019-03-20:
> This driver supports the RTL8723BU wifi part. The Bluetooth part needs a different driver. I used to provide a driver for the BT part, but that is now included as part of the btusb driver in later kernels. You will need to backport that code into the 4.6 kernel.

could you give me the BT part driver of RTL8723BU?I need it really.
Thank you for your help!

lwfinger, 2019-03-20:
The BT parts of most of the Realtek chips are the same. What I have is in http://github.com/lwfinger
/rtl8723au_bt.git. I do not support that driver, I have no idea if it works. The one in the latest kernels does work.

wwfdzh2015, 2019-03-20:
> The BT parts of most of the Realtek chips are the same. What I have is in [http://github.com/lwfinger](https://github.com/lwfinger)
> /rtl8723au_bt.git. I do not support that driver, I have no idea if it works. The one in the latest kernels does work.
Thanks a lot...
I found the linux 4.6 kernel contains Realtek usb bluetooth driver, and i config for it.
It seems that the BT part is recognized, the device infomation is:
**_zynq> hciconfig version
hci0:   Type: USB
        BD Address: 00:00:00:00:00:00 ACL MTU: 0:0 SCO MTU: 0:0
        DOWN 
        RX bytes:70 acl:0 sco:0 events:5 errors:0
        TX bytes:15 acl:0 sco:0 commands:5 errors:0_**

but i can't up the hci0, it shows that:
**_zynq> hciconfig hci0 up
Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_fw.bin failed with error -2
Bluetooth: hci0: Failed to load rtl_bt/rtl8723b_fw.bin
Can't init device hci0: No such file or directory (2)_**
However, i already copied the file rtl8723b_fw.bin to the /lib/firmware/ directory. Dose the kernel driver not support for rtl8723bu BT part?
Thank you for you help again!!