lwfinger/rtl8723bu #135: RTL8723BU Can't See 5GHZ

Poincare1, 2019-03-25:
Hello, 
I followed the auto DKMS install method and it appears to be installed correctly for the rtl8723bu.  However, it is only seeing 2.4ghz access points and not 5Ghz ones.  

Is there a method for getting this kernel module to also recognize 802.11n access points on 5ghz? 

Thank you.

lwfinger, 2019-03-25:
On my system, I get the following:

finger@linux-n6zm:~/rtl8723bu> sudo insmod 8723bu.ko
finger@linux-n6zm:~/rtl8723bu> iw dev
phy#0
        Interface wls1
                ifindex 3
                wdev 0x1
                addr 00:e0:4c:09:94:1d
                ssid Larry_ac2_5GHz
                type managed
                channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 20.00 dBm

Note that the 8723bu is connected to a 5GHz AP, thus the problem is not with the driver.

Run the command 'lsmod | grep 8xxxu'. If you see any output, then you need to unload rtl8xxxu. To prevent future problems, you should blacklist it.

Poincare1, 2019-03-25:
Not sure what the issue is.  I am not running the 8xxxu module. There is no output from lsmod | grep 8xxxu nor is it listed in the loaded modules. Again, it's not seeing any 5ghz APs.  Here is output from iwconfig:

root@rock64:~# iwconfig
lo        no wireless extensions.

wlx74da38f3cbcc  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

root@rock64:~#

lwfinger, 2019-03-25:
Use 'iw list' to see the capabilities of the device. Does it show 2 bands present?

Poincare1, 2019-03-25:
Requested output:

    root@rock64:~# iw list
    Wiphy phy0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (30.0 dBm)
			* 2472 MHz [13] (30.0 dBm)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (30.0 dBm)
			* 5180 MHz [36] (30.0 dBm)
			* 5190 MHz [38] (30.0 dBm)
			* 5200 MHz [40] (30.0 dBm)
			* 5210 MHz [42] (30.0 dBm)
			* 5220 MHz [44] (30.0 dBm)
			* 5230 MHz [46] (30.0 dBm)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (30.0 dBm)
			* 5280 MHz [56] (30.0 dBm)
			* 5300 MHz [60] (30.0 dBm)
			* 5320 MHz [64] (30.0 dBm)
			* 5500 MHz [100] (30.0 dBm)
			* 5520 MHz [104] (30.0 dBm)
			* 5540 MHz [108] (30.0 dBm)
			* 5560 MHz [112] (30.0 dBm)
			* 5580 MHz [116] (30.0 dBm)
			* 5600 MHz [120] (30.0 dBm)
			* 5620 MHz [124] (30.0 dBm)
			* 5640 MHz [128] (30.0 dBm)
			* 5660 MHz [132] (30.0 dBm)
			* 5680 MHz [136] (30.0 dBm)
			* 5700 MHz [140] (30.0 dBm)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [192] (disabled)
			* 5980 MHz [196] (disabled)
			* 6000 MHz [200] (disabled)
			* 6020 MHz [204] (disabled)
			* 6040 MHz [208] (disabled)
			* 6060 MHz [212] (disabled)
			* 6080 MHz [216] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
 	Device supports scan flush.
     root@rock64:~#

lwfinger, 2019-03-25:
That output shows that the device and mac80211 have enabled the low and middle bands of 5G. The high band channels (above 180) are disabled. On my system, those channels do not even show in the list.

Any idea what channels 5G Aps in your neighborhood are using? Also be aware that 5G penetrates a lot less than 2.4G. On one of my APs, the signals are -18 and -37 dBm for 3.4 and 5G bands.

Have you run sudo iw  dev wlx74da38f3cbcc scan | egrep "signal|SSID"'? That is an awful name for the device. Mine is "wls1".

Poincare1, 2019-03-25:
Not sure why channels above 180 are disabled.  I'm not doing anything special to either enable or disable bands or change the name. I just straight followed the procedure for DKMS install.  Would love a name better than wlx74... 

5ghz channel APs in the vicinity are on channels: 36, 40, 44, 48, 136, 149, 153,  and 157. My 5G wifi AP is on channel 157. 

In output of " iw dev wlx74da38f3cbcc scan  | egrep "signal|SSID" "  my 5G access point is not listed.

lwfinger, 2019-03-26:
Any limitations would be in the user-space networking code, not in the driver.

You might be able to control the interface name in /etc/udev/rules.d/70-persistent-net.rules.

What country is your place of residence? Is CRDA installed on your computer?

Poincare1, 2019-03-26:
I am in the USA. CRDA was not installed but it is now. Rebooted. Still not able to see any 5G WIFI APs.

lwfinger, 2019-03-26:
What distro are you using, and what method of controlling networks? I will try to duplicate your results.

Poincare1, 2019-03-27:
Using Debian Stretch with OpenMediaVault 4 for the Rock64.

lwfinger, 2019-03-27:
The ARM processor should not make any difference. Unfortunately, I do not have any ARM systems for testing. The closest is a Keezel VPN appliance, which has an embedded RTL8192DU. The %G band works fine there.

lwfinger, 2019-03-28:
I'm sorry I misled you. The RTL8723BU is 2.4G only. I had grabbed my RTL8811AU stick when I thought I had the RTL8723BU. The packages are nearly identical.

Poincare1, 2019-03-30:
Hmmm.  Well, I guess **THAT** solves the 5G problem.  📶

closed 2019-03-30 by Poincare1