lwfinger/rtl8723bu #137: "modprobe: ERROR: could not insert '8723bu': Exec format error" with "8723bu: version magic"

althio, 2019-04-08:
Hello,

The first steps seem OK on my system (new hardware, small laptop with no Linux drivers):
```
$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-17-generic/build M=/home/althio/config/wifi.driver/rtl8723bu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-17-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-17-generic'
```
```
$ sudo make install
install -p -m 644 -D 8723bu.ko /lib/modules/4.18.0-17-generic/kernel/drivers/net/wireless/8723bu.ko
/sbin/depmod  -a 4.18.0-17-generic
install rtl8723b_fw.bin -D /lib/firmware/rtl_bt/rtl8723b_fw.bin
```
but then modprobe inserting fails:
```
$ sudo modprobe -v 8723bu
insmod /lib/modules/4.18.0-17-generic/kernel/drivers/net/wireless/8723bu.ko 
modprobe: ERROR: could not insert '8723bu': Exec format error
```
with the additional info about version discrepancy:
```
$ dmesg | grep 8723
[    6.206525] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[    6.206530] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[    6.208249] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[    6.208255] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[  360.617572] 8723bu: version magic '4.18.0-10-generic SMP mod_unload ' should be '4.18.0-17-generic SMP mod_unload '
```
Is there something known, what should should check or modify?

althio, 2019-04-08:
The system was completely reinstalled, I cannot reproduce the error now.
Modprobe inserting 8723bu is OK.

closed 2019-04-08 by althio

lwfinger, 2019-04-08:
Yes, the last line stated that the kernel sources you used to build were different than the running kernel. Once you resolved that with a reinstall, that discrepancy was gone. Updating your kernel headers would have been a quicker fix.