lwfinger/rtl8723bu #138: 2 sticks simultaniously on a raspberry ?

raspi-GitHub, 2019-04-13:
hi folks,

**when configuring the 2nd EDIMAX EW-7611ulb I get**:
:~ $ sudo ifdown wlan4 && sudo ifup wlan4
ifdown: interface wlan4 not configured
RTNETLINK answers: File exists
ifup: failed to bring up wlan4
:~ $ sudo ifdown wlan3 && sudo ifup wlan3 
ifdown: interface wlan3 not configured
RTNETLINK answers: File exists
ifup: failed to bring up wlan3

**configuring the 1st stick runs well, everything runs fine**

- what's wrong with the 2nd config?
- is it at least possible to run 2 or more sticks simultaniously?

regards

lwfinger, 2019-04-13:
It may not be possible to run 2 or more. The reason is that Realtek drivers tend to use global variables too much. If those are read-only, then there is no problem, but if the driver writes to one of these, then multiple devices will fall over each other.

If you identify such globals, I will try to fix the issue. At the moment, I do not have time to look for such problems.

Lumumba, 2019-09-04:
Just wanted to state that for me with linux Mint 19.2 the rtl8723bu now works without problems with the live usb stick. So maybe the bugs with rtl8xxx drivers was fixed.
So maybe you just have to wait for a newer kernel for raspi and it will work.