lwfinger/rtl8723bu #139: trying to compile v4.3.16 on raspberry fails !!

raspi-GitHub, 2019-04-13:
hi folks,

after: git clone https://github.com/lwfinger/rtl8723bu.git -b v4.3.16

**when trying to compile branch v4.3.16. on a raspberry I get**:
:~/git/rtl8723bu-v4.3.16 $ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.98+/build M=/home/pi/git/rtl8723bu-v4.3.16  modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.14.98+“ wird betreten
  CC [M]  /home/pi/git/rtl8723bu-v4.3.16/core/rtw_cmd.o
In file included from /home/pi/git/rtl8723bu-v4.3.16/include/drv_types.h:32:0,
                 from /home/pi/git/rtl8723bu-v4.3.16/core/rtw_cmd.c:22:
/home/pi/git/rtl8723bu-v4.3.16/include/osdep_service.h: In function ‘thread_enter’:
/home/pi/git/rtl8723bu-v4.3.16/include/osdep_service.h:343:2: **error**: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  **allow_signal**(SIGTERM);
  ^~~~~~~~~~~~
/home/pi/git/rtl8723bu-v4.3.16/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/pi/git/rtl8723bu-v4.3.16/include/osdep_service.h:353:6: **error**: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (**signal_pending** (current))
      ^~~~~~~~~~~~~~
/home/pi/git/rtl8723bu-v4.3.16/include/osdep_service.h:355:3: **error**: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   **flush_signals**(current);
   ^~~~~~~~~~~~~
^Cscripts/Makefile.build:326: die Regel für Ziel „/home/pi/git/rtl8723bu-v4.3.16/core/rtw_cmd.o“ scheiterte
make[2]: *** [/home/pi/git/rtl8723bu-v4.3.16/core/rtw_cmd.o] Unterbrechung
Makefile:1535: die Regel für Ziel „_module_/home/pi/git/rtl8723bu-v4.3.16“ scheiterte
make[1]: *** [_module_/home/pi/git/rtl8723bu-v4.3.16] Unterbrechung
Makefile:320: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Unterbrechung

**:~ $ uname -a** 
Linux pi 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux

raspi-GitHub, 2019-04-13:
**additional information**: main line compiles fine !!

lwfinger, 2019-04-13:
KI pushed a fix. V4.3.16 was missing a conditional include that did not show up for x86 builds. Apparently, ARM is as little different.

closed 2019-04-13 by lwfinger