lwfinger/rtl8723bu #140: compilation -b v4.3.16 fails on raspberry - again !!

raspi-GitHub, 2019-04-14:
hi there, 

thanks for helping out so quick, but still fails - now later:

  CC [M]  /home/pi/git/tmp/rtl8723bu/os_dep/xmit_linux.o                   
  CC [M]  /home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.o                   
/home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.c: In function ‘_dynamic_check_timer_handler’:
/home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.c:137:5: **error**: ‘adapter’ undeclared (first use in this function)                                                                              
  if(**adapter**->pbuddy_adapter)
     ^~~~~~~
/home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.c:137:5: **note: each undeclared identifier is reported only once for each function it appears in**                                                
scripts/Makefile.build:326: die Regel für Ziel „/home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.o“ scheiterte
make[2]: *** [/home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.o] Fehler 1
Makefile:1535: die Regel für Ziel „_module_/home/pi/git/tmp/rtl8723bu“ scheiterte
make[1]: *** [_module_/home/pi/git/tmp/rtl8723bu] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.14.98+“ wird verlassen
Makefile:320: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2

regards,

lwfinger, 2019-04-14:
I pushed a fix that should work.

raspi-GitHub, 2019-04-19:
now I get the same problems mentioned before by another thread:

===================================================

  CC [M]  /home/pi/git/tmp/rtl8723bu/os_dep/mlme_linux.o
  CC [M]  /home/pi/git/tmp/rtl8723bu/os_dep/recv_linux.o
  CC [M]  /home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.o
/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: **error**: initialization from incompatible pointer type [-Werror=**incompatible-pointer-types**]
...........................^~~~~~~~~~               
/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: **error**: initialization from incompatible pointer type [-Werror=**incompatible-pointer-types**]
  .del_virtual_intf = **cfg80211_rtw_del_virtual_intf**,
.....................................^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:326: die Regel für Ziel „/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.o“ scheiterte
make[2]: *** [/home/pi/git/tmp/rtl8723bu/os_dep/ioctl_cfg80211.o] Fehler 1
Makefile:1535: die Regel für Ziel „_module_/home/pi/git/tmp/rtl8723bu“ scheiterte
make[1]: *** [_module_/home/pi/git/tmp/rtl8723bu] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.14.98+“ wird verlassen
Makefile:320: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2

raspi-GitHub, 2019-04-19:
:~/git/tmp/rtl8723bu $ **gcc -v**
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)

boelle, 2019-04-27:
i also get 

```
/home/pi/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]

                         ^~~~~~~~~~
/home/pi/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/pi/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .del_virtual_intf = cfg80211_rtw_del_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target '/home/pi/rtl8723bu/os_dep/ioctl_cfg80211.o' failed

```

lwfinger, 2019-04-27:
What kernel are you running? 'uname -r' will give that info.

Next run the command
`less /lib/modules/$(uname -r)/build/include/net/cfg80211.h
`
and search for "del_virtual_intf". It should look like:
```
       int     (*del_virtual_intf)(struct wiphy *wiphy,
                                    struct wireless_dev *wdev);

``````
Post what yours says.

raspi-GitHub, 2019-04-28:
I'm trying this on 2 different pi/zeros with 2 installations, which should be quite up-to-date
**this is on one of my installations** just another clone:
 . . . **git clone https://github.com/lwfinger/rtl8723bu.git -b v4.3.16 rtl-v4.3.16**

pi@zero: /git $ **cd rtl-v4.3.16**
pi@zero: /git/rtl-v4.3.16 $ **make**
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.98+/build M=/home/pi/git/rtl-v4.3.16  modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.14.98+“ wird betreten
  CC [M]  /home/pi/git/rtl-v4.3.16/core/rtw_cmd.o
 . . .
  CC [M]  /home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.o
/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.c:6361:25: error: initialization from incompatible pointer type [-**Werror=incompatible-pointer-types**]
                         **^~~~~~~~~~**               
/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.c:6361:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.c:6384:22: error: initialization from incompatible pointer type [-**Werror=incompatible-pointer-types**]
  .del_virtual_intf = **cfg80211_rtw_del_virtual_intf**,
                      **^~~~~~~~~~~~~~~~~~~~~~~~~~~~~**
/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.c:6384:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:326: die Regel für Ziel „/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.o“ scheiterte
make[2]: *** [/home/pi/git/rtl-v4.3.16/os_dep/ioctl_cfg80211.o] Fehler 1
Makefile:1535: die Regel für Ziel „_module_/home/pi/git/rtl-v4.3.16“ scheiterte
make[1]: *** [_module_/home/pi/git/rtl-v4.3.16] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.14.98+“ wird verlassen
Makefile:320: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2

pi@zero:/git/rtl8723bu-v4.3.16 $ **uname -r**
**4.14.98+**
pi@zero:/git/rtl8723bu-v4.3.16 $ **less /lib/modules/$(uname -r)/build/include/net/cfg80211.h** 

        struct wireless_dev * (*add_virtual_intf)(struct wiphy *wiphy,
                                                  const char *name,
                                                  unsigned char name_assign_type,
                                                  enum nl80211_iftype type,
                                                  struct vif_params *params);
        int     (*del_virtual_intf)(struct wiphy *wiphy,
                                    struct wireless_dev *wdev);
        int     (*change_virtual_intf)(struct wiphy *wiphy,
                                       struct net_device *dev,
                                       enum nl80211_iftype type,
                                       struct vif_params *params);

lwfinger, 2019-04-28:
I do not know what is wrong with the del_virtual_intf() call. The code in the routine matches the one in your header:
```
    int     (*del_virtual_intf)(struct wiphy *wiphy,
                                struct wireless_dev *wdev);
```
I do know what is wrong with add_virtual_intf(). The code has
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,6,0))
static struct wireless_dev *
#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,38)) || defined(COMPAT_KERNEL_RELEASE)
static struct net_device *
#else
static int
#endif
        cfg80211_rtw_add_virtual_intf(
                struct wiphy *wiphy,
        #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,7,0))
                const char *name,
        #else
                char *name,
        #endif
        #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,2,0))
                unsigned char name_assign_type,
        #endif
                enum nl80211_iftype type,
        #if (LINUX_VERSION_CODE < KERNEL_VERSION(5,0,0))
                u32 *flags,
        #endif
                struct vif_params *params)
```
Your header does not have the "u32 *flags" item that was in kernels earlier than 5.0. That means that your distro has backported a change in kernel API. If I change the repo, then I break the driver for every user whose distro has NOT made this change. I cannot do that, therefore, you will need to change it manually on your site. When Ubuntu started this abominable practice of doing this kind of backporting, I resolved that I would never try to keep up with them.

You should manually make the definition of cfg80211_rtw_add_virtual_intf() be
```
static struct wireless_dev *cfg80211_rtw_add_virtual_intf(
                struct wiphy *wiphy,
                 const char *name,
                  unsigned char name_assign_type,
                  enum nl80211_iftype type,
                  struct vif_params *params);
```
and make the definition of cfg80211_rtw_del_virtual_intf() be
```
static int cfg80211_rtw_del_virtual_intf(struct wiphy *wiphy,
        struct wireless_dev *wdev)
```
Once you make these changes, then you will not be able to do a simple 'git pull' as that will likely cause conflicts. Put the changes in a patch file that can be reverted before you go a pull and reinstated after.

closed 2019-04-28 by lwfinger

dekardkrr, 2020-01-26:
If i understand right i need to add code from previous comment  in file cfg80211.h.
I tried to paste code above line with code:
`struct cfg80211_ops {`
and i got compilation error :
`  CC [M]  /opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.o
/opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:4023:2: error: conflicting types for ‘cfg80211_rtw_add_virtual_intf’
  cfg80211_rtw_add_virtual_intf(
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/8723/rtl8723bu/include/osdep_service_linux.h:81,
                 from /opt/8723/rtl8723bu/include/osdep_service.h:41,
                 from /opt/8723/rtl8723bu/include/drv_types.h:32,
                 from /opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:3131:29: note: previous declaration of ‘cfg80211_rtw_add_virtual_intf’ was here
 static struct wireless_dev *cfg80211_rtw_add_virtual_intf(
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, enum nl80211_iftype,  struct vif_params *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, enum nl80211_iftype,  u32 *, struct vif_params *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, enum nl80211_iftype,  unsigned int *, struct vif_params *)’} [-Werror=incompatible-pointer-types]
  .add_key = cfg80211_rtw_add_key,
                         ^~~~~~~~~                
/opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:6361:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: error: initialization of ‘struct wireless_dev * (*)(struct wiphy *, const char *, unsigned char,  enum nl80211_iftype,  struct vif_params *)’ from incompatible pointer type ‘struct wireless_dev * (*)(struct wiphy *, const char *, unsigned char,  enum nl80211_iftype,  u32 *, struct vif_params *)’ {aka ‘struct wireless_dev * (*)(struct wiphy *, const char *, unsigned char,  enum nl80211_iftype,  unsigned int *, struct vif_params *)’} [-Werror=incompatible-pointer-types]
  .del_virtual_intf = cfg80211_rtw_del_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.c:6384:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:304: /opt/8723/rtl8723bu/os_dep/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1522: _module_/opt/8723/rtl8723bu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.93-v7+'
make: *** [Makefile:320: modules] Error 2
`
I tried to paste this code below that line and i got error too.
Could you show the place where need to paste this code?

lwfinger, 2020-01-26:
Why do you think you need to add any code? If you do a build with nothing added, what happens? I think I have the definitions of the kernel routines done correctly. The only thing that might be wrong happens when your distro backports kernel APIs to older kernels. I DO NOT support that awful practice, and I will not help builds that suffer from such idiotic "features".