lwfinger/rtl8723bu #144: New base driver version v5.2.17.1

masbug, 2019-06-05:
Hi,

there seems to be a new base driver available: https://github.com/thisiswangle/WGD01CM3-rtl8723bu
Thought you'd want to know :)

Kind regards,
M.

EdKee, 2019-07-15:
Thank you very much for the link for v5.2.17.1. Is it original driver from Realtek? If not, how could I look at untouched Realtek driver?

I'm new to the Realtek Linux drivers zoo. V5 driver seems improving AP mode a lot. 1) V4 drivers in AP mode can easily make RTW_CMD_THREAD sticking to high CPU usage. 2) As well there were various problems connecting WiFi clients using V4 drivers. Android 8 phone for some reason was disconnecting each couple of seconds, while Android 9 and 4 phones and Windows 10 client were operating. 3) As well AP hotspot could start giving no internet connection, unsticking sometimes happened after few minutes of disconnection, hard to predict.
V4.4.5 driver for LM811 module (https://www.lm-technologies.com/product/wifi-and-bluetooth-usb-module-4-0-dual-mode-class-1-lm811/) seems being better in AP mode than the driver from this git (https://github.com/lwfinger/rtl8723bu), but still bad. Hope BT function will work as well. 

Regards,
E

lwfinger, 2019-07-15:
Why are you asking here for a driver published in another repo?

closed 2019-07-15 by lwfinger

EdKee, 2019-07-16:
Sorry. First of all thank you very much for your repo. As well thanks for masbug for the link, which most likely I would ignore if it didn't appear here. I would be happy with your repo even if it was v2 of Realtek driver, of course if it worked properly as well in AP mode. I'd love to see any repo which would contain working driver. I suspected I'll ask too much, but please understand my frustration, it is totally not clear where from these repos come from, where is the good source for information about Realtek drivers since they don't provide it? Any  news or something?

lwfinger, 2019-07-16:
The drivers ALL come from Realtek. On the original form, the drivers spam the logs mercilessly and do not build on newer kernels! The various GitHub repos arise because different people decide to provide some support for them. Through my long-term support of Realtek PCI drivers in the drivers/net/wireless/realtel/rtlwifi tree of the mainline kernel, I have better access to Realtek drivers than most people do, thus I can get whatever drivers I want. Thus in my case, If I choose not to implement a new version, it is because I do not have time nor the inclination to take on the extra work. I am a volunteer and I control how much effort I put in. The advantage is that I can refuse to do anything.

EdKee, 2019-07-17:
Thanks for your repo I learned how to port them to new kernel version, as well that it seems no one cares about new kernels compatibility with the older. What's not clear is that on Realtek.com nearly all chips have only drivers for Windows..  

Best Regards,
Edward