lwfinger/rtl8723bu #148: Build fail because of structleak_plugin

Brandon2255p, 2019-07-24:
HI there
Thank you so much for making this driver available. It was working until my kernel updated to 5.2.2 
using gcc version 9.1.1 20190721 gcc-9-branch@273639 (Clear Linux OS for Intel Architecture) 

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.2.2-802.native/build M=/home/brandon/wifi  modules
make[1]: Entering directory '/usr/lib/modules/5.2.2-802.native/build'
  CC [M]  /home/brandon/wifi/core/rtw_cmd.o
cc1: error: incompatible gcc/plugin versions
cc1: error: fail to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:279: /home/brandon/wifi/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1595: _module_/home/brandon/wifi] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.2.2-802.native/build'
make: *** [Makefile:333: modules] Error 2
```

Brandon2255p, 2019-07-24:
@lwfinger I have tried for a few hours to figure out what is preventing this build. Since I had no luck, I was hoping you may have some more insight?

lwfinger, 2019-07-24:
That file is an optional part of the kernel. The error is that the version of gcc on your system differs from the one used to compile the kernel. You need to upgrade your gcc to version 9.1.1 or get the kernel source and recompile the kernel.

I looked around on the Clear Linux web site, and could only find gcc 7 and gcc 8 packages. I would complain if they are building a kernel with a compiler that they do not furnish.

closed 2019-07-24 by lwfinger

Brandon2255p, 2019-07-25:
@lwfinger thank you very much for that insight. There is now an open issue on the clear linux project and they have recognised that they should have rebuilt their kernels when they updated gcc