lwfinger/rtl8723bu #149: sudo modprobe -v 8723bu; Operation not permitted

dagohil, 2019-08-17:
I try to get my WLAN USB stick (Edimax EW 7611 ULB up and running by following your instructions to install the driver 8723 bu. Everything works fine, but when doing "sudo modprobe -v 8723bu" I get the following message: "insmod /lib/modules/5.0.0-25-generic/kernel/drivers/net/wireless/8723bu.ko modprobe: ERROR: could not insert '8723bu': Operation not permitted".
My Laptop is running Ubuntu Mate 19.04.

Thanks in advance

Hilmar

lwfinger, 2019-08-19:
Do you have secure boot enabled? As I cannot sign the module, that option must be disabled in the BIOS to be able to use an external module.

dagohil, 2019-08-21:
Yes, unfortunately this is the answer. Unfortunately as I could only get Ubuntu up and running was with secure boot enabled :) New WLan module ordered.

closed 2019-08-21 by dagohil

lwfinger, 2019-08-22:
If Ubuntu will not let you get the system running with secure boot disabled, then you need to find a new distro. That is preposterous, but just exactly what I would expect.