lwfinger/rtl8723bu #150: DKMS Compile Fail On OdroidN2 DebianArmbian

Poincare1, 2019-09-15:
Compile fails on the following step from DKMS install: 

`sudo dkms autoinstall $PACKAGE_NAME/$PACKAGE_VERSION

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=4.9.190-odroidn2 -C /lib/modules/4.9.190-odroidn2/build M=/var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.190-odroidn2 (aarch64)
Consult /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/make.log for more information.
`

And, here is the output of /var/lib/.../build/make.log: 

`DKMS make.log for rtl8723bu-4.3.6.11_12942.20141204_BTCOEX20140507-4E40 for kernel 4.9.190-odroidn2 (aarch64)
Sun Sep 15 15:50:50 CDT 2019
make: Entering directory '/usr/src/linux-headers-4.9.190-odroidn2'
  LD      /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/built-in.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_set.o
In file included from ./include/linux/irqflags.h:15,
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_security.c:22:
./arch/arm64/include/asm/irqflags.h:24:10: fatal error: ../drivers/amlogic/debug/irqflags_debug_arm64.h: No such file or directory
 #include <../drivers/amlogic/debug/irqflags_debug_arm64.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from ./include/linux/irqflags.h:15,
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_debug.c:22:
./arch/arm64/include/asm/irqflags.h:24:10: fatal error: ../drivers/amlogic/debug/irqflags_debug_arm64.h: No such file or directory
 #include <../drivers/amlogic/debug/irqflags_debug_arm64.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_security.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from ./include/linux/irqflags.h:15,
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_io.c:48:
./arch/arm64/include/asm/irqflags.h:24:10: fatal error: ../drivers/amlogic/debug/irqflags_debug_arm64.h: No such file or directory
 #include <../drivers/amlogic/debug/irqflags_debug_arm64.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_debug.o] Error 1
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_io.o] Error 1
In file included from ./include/linux/irqflags.h:15,
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_cmd.c:22:

Sun Sep 15 15:50:50 CDT 2019
make: Entering directory '/usr/src/linux-headers-4.9.190-odroidn2'
  LD      /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/built-in.o
  CC [M]  /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_cmd.o
[... 49 lines cut ...]
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_query.c:22:
./arch/arm64/include/asm/irqflags.h:24:10: fatal error: ../drivers/amlogic/debug/irqflags_debug_arm64.h: No such file or directory
 #include <../drivers/amlogic/debug/irqflags_debug_arm64.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_cmd.o] Error 1
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_query.o] Error 1
In file included from ./include/linux/irqflags.h:15,
                 from ./include/linux/spinlock.h:53,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service_linux.h:24,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/osdep_service.h:40,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_set.c:22:
./arch/arm64/include/asm/irqflags.h:24:10: fatal error: ../drivers/amlogic/debug/irqflags_debug_arm64.h: No such file or directory
 #include <../drivers/amlogic/debug/irqflags_debug_arm64.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build/core/rtw_ioctl_set.o] Error 1
make: *** [Makefile:1628: _module_/var/lib/dkms/rtl8723bu/4.3.6.11_12942.20141204_BTCOEX20140507-4E40/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.9.190-odroidn2'`

lwfinger, 2019-09-16:
I do not know enough about ARM to tell you what is wrong, other than to say that your kernel header package is incomplete. You need to contact an ARM kernel forum to learn why.

dev-0x7C6, 2019-09-16:
Maybe this will help: https://forum.odroid.com/viewtopic.php?t=34659

Roker72, 2019-11-28:
Удали firmware-rtl8723bu можешь через synaptic после чего установи dkms после этого собери драйвер и будет тебе счастье метод рабочий сборку делай по инструкции 1 шаг второй и последний после перегружаешься и всё заработает