lwfinger/rtl8723bu #152: Double entry in Networkmanager and no reconnect after suspend

dagohil, 2019-10-04:
Hello,

I followed your instructions to get the 8723bu driver installed. I also blocked the RTL 8xxxu. 

In my Networkmanager I have two entries both named Realtek Wlan. Both work both the driver 8723bu. One is working fine, the other looks fine but does not really work (or very slow). Bluetooth is working also fine. After suspend I cant connect to none of the entries and bluetooth is not working. I did not cancel concurrence mode. Any ideas?

lwfinger, 2019-10-04:
CONFIG_CONCURRENT_MODE is turned ON in the standard system. That is the reason you get two interfaces. If you do not want them, then change Makefile.

As I have stated many, many, many times, Bluetooth drivers are provided as a convenience. I do not debug them.

I do not know why the device fails after suspend.

lwvmobile, 2019-12-30:
I was facing a similar issue with the the network interface not reconnecting on resume. I did some tinkering in the Makefile to no avail by changing the following options:

CONFIG_USB_AUTOSUSPEND = n
CONFIG_HW_PWRP_DETECTION = n

I changed these two options from n to y as my device is listed as a USB device. I believe that is the way it is manufactured, as a USB device as opposed to directly on the bridge or PCIE or whatever. 

CONFIG_USB_AUTOSUSPEND = y
CONFIG_HW_PWRP_DETECTION = y

Then I did the following:

sudo modprobe -r 8723bu
sudo make uninstall 
make
sudo make install
sudo modprobe -v 8723bu

Sadly, these changes didn't fix the issue, but I did find a simple workaround.

After resuming, simply run:
sudo modprobe -r 8723bu
sudo modprobe -v 8723bu

That will get the interface back up. 

Optionally, just make a simple bash script to do it for you:

sudo modprobe -r 8723bu
sleep 5
sudo modprobe -v 8723bu

evgen-y2k, 2020-04-02:
I  solved this problem using script that will be executed after suspend. Script:
``

Lumumba, 2020-04-25:
You two have blacklisted rtl8xxxu Kernel driver?

kevin201, 2020-10-14:
I was also affected by the not connecting after suspend problem. Solved by the solution here, https://askubuntu.com/questions/1029250/ubuntu-18-04-ethernet-disconnected-after-suspend and above, with the exception that I had to use

modprobe rtl8723bu

to insert the module again.

This was on Ubuntu 20.04 kernel 5.4.0-51-generic #56-Ubuntu SMP

 dkms status
rtl8723bu, 4.3.6.11_12942.20141204_BTCOEX20140507-4E40, 5.4.0-48-generic, x86_64: installed
rtl8723bu, 4.3.6.11_12942.20141204_BTCOEX20140507-4E40, 5.4.0-51-generic, x86_64: installed

lsusb|grep -i real
Bus 002 Device 008: ID 0bda:b720 Realtek Semiconductor Corp. 802.11n WLAN Adapter

```
$ cat /lib/systemd/system-sleep/rtl8723bu-refresh
#!/bin/bash

PROGNAME=$(basename "$0")
state=$1
action=$2

function log {
    logger -i -t "$PROGNAME" "$*"
}

log "Running $action $state"

if [ "${1}" == "post" ]; then
    rmmod 8723bu \
    && log "Removed rtl8723bu" \
    && modprobe rtl8723bu \
    && log "Inserted rtl8723bu"
fi
```

ivnish, 2021-04-06:
@kevin201 thank you, it helps!