lwfinger/rtl8723bu #153: Driver not working after kernel upgrade

dagohil, 2019-10-06:
Hello,
After upgrading the kernel and therefore repeating the installation procedure, I got the following error message:

Dago@dago-MYBOOK-14:~/rtl8723bu$ sudo modprobe -v 8723bu
insmod /lib/modules/5.0.0-31-generic/kernel/drivers/net/wireless/8723bu.ko 
modprobe: ERROR: could not insert '8723bu': Exec format error

Help appreciated

Hilmar

lwfinger, 2019-10-07:
When these kinds of errors occur, you need to check the output of the dmesg command to get a better idea of the cause.

dagohil, 2019-10-07:
Thanks in advance for your assistance.

This is a very frequent msg and the only one related to 8723bu

8723bu: version magic '5.0.0-29-generic SMP mod_unload ' should be '5.0.0-31-generic SMP mod_unload '

Hilmar

lwfinger, 2019-10-07:
Your kernel header version do not match that of the running kernel. The 'uname -r' will tell you the kernel version.

dagohil, 2019-10-07:
Yes I understand my new kernel is 5.0.0-31-generic and the header is 5.0.0-29. How do get them to match? Would be "sudo apt-get install linux-headers-generic" the right command. And if so what next? Sorry, but I dont know exactly what I am doing therefore I am a little bit cautious. :)

Hilmar

lwfinger, 2019-10-07:
You should not be able to hurt anything by installing kernel headers; however, your distro's forum would be a better place to ask. My distro does not use apt!

closed 2019-10-07 by lwfinger