lwfinger/rtl8723bu #154: Make error on fedora 30

mefiz, 2019-11-05:
Upon running make I get the following error:
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.9-301.fc30.x86_64/build M=/home/Biscuits/rtl8723bu  modules
make[1]: *** /lib/modules/5.0.9-301.fc30.x86_64/build: No such file or directory.  Stop.
make: *** [Makefile:333: modules] Error 2`

I can confirm kernel dev and header files packages are installed.

lwfinger, 2019-11-05:
Well, make says they are not there. Do
ls /lib/modules/5.0.9-301.fc30.x86_64/build
to confirm. Does 'uname -r' match that "5.0.9-301.fc30.x86_64"?

mefiz, 2019-11-06:
uname-r matches. However /build is a directory that does not exist.

lwfinger, 2019-11-06:
Then you have not completed the setup. Fedora documentation says that you need kernel-devel and kernel-headers. Use
su -c 'yum install kernel-devel kernel-headers'
I am not a Fedora user and have no idea if that is all you need.

mefiz, 2019-11-06:
Fedora says packages are installed. Probably the issue lies somewhere else. Thanks though.

closed 2019-11-06 by mefiz

lwfinger, 2019-11-06:
When you find what caused this issue, please add the solution here. I will see it and be better able to help the next Fedora user with this problem.