lwfinger/rtl8723bu #155: rtl8xxxu on Presigio Smartbook 133s finally works (almost)

cypa, 2019-11-07:
Solution is next:

- boot smartbook
- check if WiFi is connection successfully
- if no connection then suspend smartbook for a moment
- wake up smartbook and check WiFi connectoin
- if still no connection then suspend again and check until WiFi connection would by successfull

cypa, 2019-11-07:
you can use RTC wakeup timeout to suspend/resume automatically

Lumumba, 2019-11-24:
This is the way the kernel driver behaves.
The rtl8723bu driver from here works on boot and on suspend.

cypa, 2019-11-24:
> This is the way the kernel driver(rtl8xxxu) behaves since like 2 years.
> The rtl8723bu driver from here works on boot and on suspend.
> 
> What Kernel version you are running?

latest Arch linux kernel (both mainline and TLS)

which device do you use? mine is Presigio Smartbook 133s

Lumumba, 2019-11-24:
I have bought 4 Medion Akoya S4019 for my neighbors.
I reported it to Ubuntu-Devs but nothing happened.
6 Months ago the rtl8xxxxu driver got an update which improved this but still you cannot use bluetooth and wlan at the same time.

cypa, 2019-11-24:
Please provide dmesg and other data on the bug on your devices.

cypa, 2019-11-24:
I've sent my data to linux-wireless@vger.kernel.org and was suggested to check https://bugzilla.kernel.org/show_bug.cgi?id=202541