lwfinger/rtl8723bu #156: README.md: does not clarify why and how of this external, non-mainline-kernel driver

andim2, 2019-11-12:
Hello,

while the README.md does mention blacklisting of the mainline kernel driver module, it fails to list potential reasons for using (and also: creating!) this implementation code which is an external, not-[yet-]mainline driver.

Thank you!

lwfinger, 2019-11-12:
This code will NEVER be in mainline. Why? It would take over a year of full-time work to rework it to reach the quality necessary. If you are willing to fund the effort, then I might consider it.

This driver was created by Realtek, the vendor of the chip. Their USB drivers "work" with Windows, FreeBSD, and Linux. As such, they contain a lot of code that is useless in Linux, but I "publish" them at GitHub so that people like you can have access to them. If you do not like what you see, you have two options - (1) Write your own driver, or (2) Update this one. I consider all suggestions that include actual code.

closed 2019-11-12 by lwfinger

andim2, 2019-11-13:
Sorry, I had written this issue in a somewhat demanding, impolite manner, not unlikely because directly prior to this I was miffed about very questionable support status (in all newer versions of stable mainline kernel prior to v5.4-rcX!) of Not-so-Realtek hardware:
- "[PATCH] rtl8xxxu: Fix wifi low signal strength issue of RTL8723BU" https://www.spinics.net/lists/netdev/msg581689.html
  (if such more-than-basic shenanigans turn out to be required after a full 3 years (2016+) of internet references of this hardware _model_ _variant_, then this is a Clear Red Flag about any slightly more peripheral aspects - say multicast or whatever - of this vendor's "nice" hardware + support combo, IOW it is not to be used for any production-type / sufficiently diverse scenarios)
- https://www.phoronix.com/forums/forum/hardware/general-hardware/1096145-realtek-contributes-new-rtw88-802-11ac-wifi-driver-to-the-linux-kernel
  "The entire support around RealTek WiFi has been messy for years."

> (1) Write your own driver

I have written my own driver ;) (albeit for original TI WLAN crap - acx100, which could be said to have been in an even more dire situation)

In closing, thank you very much for your very appreciable and major efforts of reducing the gap between what ought to be vendor-provided as a sufficiently robust, stable service and what all too often actually happens!

lwfinger, 2019-11-13:
My only comment on the above rant is that rtl8xxxu was created without any Realtek input other than this external (crappy) driver. Yes, the patch quoted above fixed an error in the transcription; however, the author of the patch needed considerable effort to find that fix. Obviously, no one had devoted that effort, which is why it took 3 years to fix it. I'm not sure what Red Flag that indicates.