lwfinger/rtl8723bu #157: [WORKAROUND] Continuous disconnects on Kernel 5.0.0-38

ziSo12, 2020-02-08:
There is an issue using Edimax EW-7611ULB wifi/bt stick on Ubuntu 19.04 Kernel 5.0.0-38.
Wifi is constantly disconnecting/reconnecting and sometimes not coming back.
Here is my current workaround. Thought this might be a power saving issue so I set
`CONFIG_POWER_SAVING = n`
I also commented out
#EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE

Source from branch v4.3.16 giving compile errors, source from branch master compiling fine, though. At least I got a stable connection so far.

aldemaroc, 2020-03-14:
I was also unable to compile v4.3.16.

lwfinger, 2020-03-14:
Without someone posting the errors, I will never fix them.

ivnish, 2021-04-06:
Thank you @ziSo12, it helps!

ejmarkow, 2021-10-16:
I had the same issue on my openSUSE Linux system. It was consistently connecting/disconnecting.

Recompiled using:
CONFIG_POWER_SAVING = n

and Commenting out
#EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE

My Wifi hotspot and Bluetooth all working well, thanks for this tip!