lwfinger/rtl8723bu #163: RTL8723bu With Concurrent Mode Build Results in Internet Connection Drops

desmotronic, 2020-04-22:
The RTL8723bu driver when built in concurrent mode causes the internet connectivity to drop.  This happens when the wireless card is solely connected to an AP.  

To solve this, the 8723bu driver needs to be build by commenting out the concurrent mode option in the MakeFile.

I have noticed this in all Debian based distributions that come shipped with this driver as it is build with concurrent mode enabled by default.  It is probably an old bug, but just wanted to report it as I always have to rebuild this driver when I do a fresh OS installation.

Thank you.

jmbott, 2020-04-29:
I experienced this same issue of the wifi dropping out. Commenting out `EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE`  and rebuilding as described seems to have corrected the problem for now. Using this driver on the Raspberry Pi 3+ compute module (Raspbian 4.19) with an Edimax EW-7611ULB wifi dongle.

ivnish, 2021-04-06:
Thank you @jmbott, it helps!