lwfinger/rtl8723bu #164: hostapd setup error

iamxinnian, 2020-05-12:
Configuration file: /etc/rtl_hostapd_2G.conf
HT (IEEE 802.11n) with WPA/WPA2 requires CCMP to be enabled, disabling HT capabilities
drv->ifindex=5
l2_sock_recv==l2_sock_xmit=0x0x62638
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=14 freq=2484 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=52 freq=5260 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=56 freq=5280 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=60 freq=5300 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=64 freq=5320 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=100 freq=5500 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=104 freq=5520 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=108 freq=5540 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=112 freq=5560 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=116 freq=5580 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=120 freq=5600 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=124 freq=5620 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=128 freq=5640 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=132 freq=5660 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=136 freq=5680 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=140 freq=5700 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=0 dBm
Scan for neighboring BSSes prior to enabling 40 MHz channel
Failed to request a scan of neighboring BSSes
HT40: control channel: 6  secondary channel: 10
Completing interface initialization
Mode: IEEE 802.11g  Channel: 6  Frequency: 2437 MHz
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
Flushing old station entries
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=0)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=1)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=2)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan0 keyidx=3)
Using interface wlan0 with hwaddr 00:13:ef:f1:11:ac and ssid 'glut02522'
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=9):
     67 6c 75 74 30 32 35 32 32                        glut02522       
PSK (ASCII passphrase) - hexdump_ascii(len=8): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
WPS: Use configured UUID - hexdump(len=16): 12 34 56 78 9a bc de f0 12 34 56 78 9a bc de f0
WPS: Build Beacon IEs
WPS:  * Version (hardcoded 0x10)
WPS:  * Wi-Fi Protected Setup State (2)
WPS: Build Probe Response IEs
WPS:  * Version (hardcoded 0x10)
WPS:  * Wi-Fi Protected Setup State (2)
WPS:  * Response Type (3)
WPS:  * UUID-E
WPS:  * Manufacturer
WPS:  * Model Name
WPS:  * Model Number
WPS:  * Serial Number
WPS:  * Primary Device Type
WPS:  * Device Name
WPS:  * Config Methods (18c)
WPS:  * RF Bands (1)
rtl871x_set_beacon_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_hidden_ssid ignore_broadcast_ssid:0, glut02522,9
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_acl
Failed to set beacon head/tail or DTIM period
rtl871x_set_wps_assoc_resp_ie
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to initialize WPS Registrar
wlan0: Unable to setup interface.
Flushing old station entries
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rmdir[ctrl_interface]: No such file or directory
ELOOP: remaining socket: sock=6 eloop_data=0x626a8 user_data=(nil) handler=0x28645

lwfinger, 2020-05-12:
Not supported by this driver.

closed 2020-05-12 by lwfinger

iamxinnian, 2020-05-13:
Are there any wpa_supplicant and hostapd software that supports the master branch?