lwfinger/rtl8723bu #165: Cannot install the driver

Lozeonjyu, 2020-06-05:
Hi,

I bought a Wifi dongle EW-7611ULB and was trying to install the driver but failed. It seems like the Linux header is required but it won't allow me to install the Linux header. Please check below:
 
```
debian@IZAR-8f704d:~/rtl8723bu$ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.8.13-bone86/build M=/home/debian/rtl8723bu modules
make: *** /lib/modules/3.8.13-bone86/build: No such file or directory. Stop.
make: *** [modules] Error 2
debian@IZAR-8f704d:~/rtl8723bu$ sudo apt-get install linux-headers-`uname -r`
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-3.8.13-bone86
E: Couldn't find any package by regex 'linux-headers-3.8.13-bone86'
debian@IZAR-8f704d:~/rtl8723bu$
```

Thanks

lwfinger, 2020-06-05:
Kernel 3.8.13 is really old - even old enough that your distro (Debian) has dropped it from its repositories. I know that upgrading for non-x86 platforms can be difficult. What ARM system do you have?

RobertCNelson, 2020-06-05:
Hi @Lozeonjyu , sorry it looks like in 2017.12.02 i stopped building for wheezy, that last version still in our BeagleBoard.org repo is bone84.

http://repos.rcn-ee.net/debian/pool/main/l/linux-upstream/linux-headers-3.8.13-bone84_1wheezy_armhf.deb

Please upgrade to the http://beagleboard.org/latest-images, @lwfinger driver is pre-installed. ;)

Regards,

Lozeonjyu, 2020-06-05:
> Kernel 3.8.13 is really old - even old enough that your distro (Debian) has dropped it from its repositories. I know that upgrading for non-x86 platforms can be difficult. What ARM system do you have?

Hi iwfinger,

Please check below:

```
debian@IZAR-8f704d:~$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 596.48
Features        : swp half thumb fastmult vfp edsp thumbee neon vfpv3 tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc08
CPU revision    : 2

Hardware        : Generic AM33XX (Flattened Device Tree)
Revision        : 0000
Serial          : 0000000000000000
```

lwfinger, 2020-06-05:
So an upgrade to the latest images should solve your problems. Good luck.