lwfinger/rtl8723bu #166: make error

lucas-mior, 2020-06-06:
```
lucas@Dell rtl8723bu $ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.12-gnu-1/build M=/home/lucas/.local/sourcecode/rtl8723bu  modules
make[1]: Entrando no diretório '/usr/lib/modules/5.6.12-gnu-1/build'

  CC [M]  /home/lucas/.local/sourcecode/rtl8723bu/core/rtw_cmd.o
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:268: /home/lucas/.local/sourcecode/rtl8723bu/core/rtw_cmd.o] Erro 1
make[1]: *** [Makefile:1683: /home/lucas/.local/sourcecode/rtl8723bu] Erro 2
make[1]: Saindo do diretório '/usr/lib/modules/5.6.12-gnu-1/build'

make: *** [Makefile:333: modules] Erro 2
```
I'm using parabola.

lucas-mior, 2020-06-06:
I changed the kernel to the ```linux-libre-lts``` (also had to install ```linux-libre-lts-headers```) and ```make``` worked!

closed 2020-06-06 by lucas-mior

lwfinger, 2020-06-06:
I was about to suggest that you rebuild your kernel with Parabola. Your kernel configuration file included the gcc plugins; however, they were an incorrect version. Updating the kernel and the headers was the correct move.