lwfinger/rtl8723bu #169: Cannot run dirver on OpenWRT 19.07 MT7688

bigdolphin, 2020-09-03:
Hello,
I am trying to implement the driver into MT7688 with OpenWRT 19.07. I successfully compiled the driver but cannot run it, everytime I insmod or modprobe the module, the system says:

> [  179.297161] RTL871X: module init start
> [  179.300977] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
> [  179.308113] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
> [  179.315968] ------------[ cut here ]------------
> [  179.320796] WARNING: CPU: 0 PID: 4777 at backports-4.19.137-1/net/wireless/core.c:743 wiphy_register+0x2a4/0x9d0 [cfg80211]
> [  179.332173] Modules linked in: 8723bu(+) rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib rt2500usb qcserial pppoe ppp_async option mt76x0u mt76x0_common ipw cdc_mbim w1_therm uvcvideo usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 snd_usb_audio smsc95xx sierra_net sierra rtlwifi rtl8187 rt73usb rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_synctty ppp_generic plusb pl2303 p54usb p54common oti6858 nf_conntrack_ipv6 mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7603e mt7601u mt76_usb mt76 mos7840 mos7720 mmc_spi mct_u232 mcs7830 mac80211 keyspan kalmia iptable_nat ipt_REJECT ipt_MASQUERADE io_edgeport huawei_cdc_ncm garmin_gps ftdi_sio dm9601 cypress_m8 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a at25 at24 asix ark3116 xt_time xt_tcpudp
> [  179.403925]  xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xr_usb_serial_common wire visor videobuf2_v4l2 vhci_hcd usbserial usbnet usblp usbip_host usbip_core usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tpm_rng spidev snd_usbmidi_lib slhc rtl8150 rng_core rfcomm r8712u r8152 pwm_mediatek_ramips pegasus of_mmc_spi nvmem_core nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache
> [  179.475823]  lzo libcrc32c kaweth iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip_tables hwmon hso hidp hid_generic hci_uart ezusb dmx_usb crc7 crc_itu_t crc_ccitt compat cdc_wdm btusb btintel bnep bluetooth snd_soc_wm8960 regmap_i2c snd_soc_simple_card snd_soc_simple_card_utils snd_soc_ralink_i2s fuse act_connmark nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred videobuf2_vmalloc videobuf2_memops videobuf2_core hid v4l2_common videodev evdev i2c_mt7621 snd_soc_core gpio_pcf857x ralink_gdma virt_dma i2c_dev i2c_core xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip
> [  179.548530]  ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ifb snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore input_core vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 dma_shared_buffer ecdh_generic sha256_generic md5 kpp ecb cmac crypto_acompress mmc_block usb_storage sdhci_pltfm sdhci mmc_core leds_gpio uhci_hcd ohci_platform ohci_hcd ehci_pci ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat btrfs zstd_decompress zstd_compress zlib_inflate zlib_deflate xxhash xor raid6_pq lzo_decompress
> [  179.620926]  lzo_compress usbcore nls_base usb_common crc16 mii tpm aead crypto_null cryptomgr crc32c_generic crypto_hash
> [  179.632119] CPU: 0 PID: 4777 Comm: insmod Not tainted 4.14.193 #0
> [  179.638314] Stack : 804b0000 80469d5c 00000000 00000000 80441aa0 85fa9a74 857cc2dc 80495947
> [  179.646862]         8043dae8 000012a9 805f3660 000002e7 87ff0400 00000001 85fa9a28 fd0ec6b0
> [  179.655389]         00000000 00000000 805f0000 00007cc0 00000000 0000018d 00000007 00000000
> [  179.663892]         00000000 804a0000 0009a537 00000000 80000000 00000009 00000000 86774520
> [  179.672381]         8674057c 000002e7 87ff0400 00000028 00000000 80210cd0 00000000 805f0000
> [  179.680880]         ...
> [  179.683375] Call Trace:
> [  179.685881] [<8000af90>] show_stack+0x58/0x100
> [  179.690392] [<80026990>] __warn+0xe4/0x13c
> [  179.694565] [<80026a78>] warn_slowpath_null+0x1c/0x34
> [  179.699796] [<8674057c>] wiphy_register+0x2a4/0x9d0 [cfg80211]
> [  179.706340] [<8355392c>] rtw_wdev_alloc+0x1a8/0x2e4 [8723bu]
> [  179.713211] [<83540a94>] rtw_set_hal_ops+0x350/0x550 [8723bu]
> [  180.129843] ---[ end trace 9cbf6cb6f031612a ]---
> [  180.134800] usbcore: registered new interface driver rtl8723bu
> [  180.140720] RTL871X: module init ret=0

These are makefile files for compiling under OpenWRT SDK:

**Path: package/kmod-rtl8723bu/Makefile**

```
include $(TOPDIR)/rules.mk
include $(INCLUDE_DIR)/kernel.mk

# Name and release number of this package
PKG_NAME:=kmod-rtl8723bu
PKG_VERSION:=1.0.0
PKG_RELEASE:=1

# This specifies the directory where we're going to build the program.
# The root build directory, $(BUILD_DIR), is by default the build_mipsel
# directory in your OpenWrt SDK directory
PKG_BUILD_DIR := $(BUILD_DIR)/$(PKG_NAME)

include $(INCLUDE_DIR)/package.mk

# Specify package information for this program.
# The variables defined here should be self explanatory.
define Package/$(PKG_NAME)	  	
	CATEGORY:=Kernel modules
	SUBMENU:=Wireless Drivers
	TITLE:=Realtek RTL8723BU Driver
	DEPENDS:=+libstdcpp +kmod-cfg80211 +kmod-usb-core
endef

define Package/$(PKG_NAME)/description
 If you can't figure out what this program does,
 you're probably brain-dead and need immediate
 medical attention.
endef

# Specify what needs to be done to prepare for building the package.
# In our case, we need to copy the source files to the build directory.
# This is NOT the default.  The default uses the PKG_SOURCE_URL and the
# PKG_SOURCE which is not defined here to download the source from the web.
[... 404 lines cut ...]
uninstall:
	rm -f $(MODDESTDIR)$(MODULE_NAME).ko
	$(DEPMOD) -a ${KVER}
	rm -f $(FW_DIR)/rtl8723b_fw.bin

config_r:
	@echo "make config"
	/bin/bash script/Configure script/config.in

.PHONY: modules clean

clean:
	cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd */*.ko
	cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
	cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
	cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
	rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
	rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
	rm -fr .tmp_versions .cache.mk
	rm convert_firmware
endif
```
Please help me solve it, I am being in stuck of this issue for months.
Thank you very much.
Best regards.

lwfinger, 2020-09-03:
Please check the code in the kernel at backports-4.19.137-1/net/wireless/core.c  line 743 to see what warning is being triggered. Once we know that, perhaps we will know how to fix it.

bigdolphin, 2020-09-04:
Thank you for your help,
I see that there are two warnings on dmesg output:

> 
> WARNING: CPU: 0 PID: 1646 at backports-4.19.137-1/net/wireless/core.c:743 wiphy_register+0x2a4/0x9d0 [cfg80211]
> WARNING: CPU: 0 PID: 1646 at backports-4.19.137-1/net/wireless/core.c:821 wiphy_register+0x61c/0x9d0 [cfg80211]

If I do not use CONFIG_IOCTL_CFG80211, I can load the module in OpenWRT and dmesg show wlan2 for this module, however, iw does not see it then I cannot scan any wifi.

This is the dmesg output when not using CONFIG_IOCTL_CFG80211:

> 
> [  140.800350] RTL871X: module init start
> [  140.804248] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
> [  140.811305] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
> [  140.976836] RTL871X: rtw_ndev_init(wlan1)
> [  140.982461] RTL871X: rtw_ndev_init(wlan2)
> [  140.988317] usbcore: registered new interface driver rtl8723bu
> [  140.994309] RTL871X: module init ret=0
> [  174.729178] RTL871X: RTW_ADAPTIVITY_EN_
> [  174.729204] AUTO, chplan:0x20, Regulation:3,3
> [  174.737563] RTL871X: RTW_ADAPTIVITY_MODE_
> [  174.737567] NORMAL
> [  175.459655] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready

And this the core.c code in the backports:

```

1. /*
2.  * This is the linux wireless configuration interface.
3.  *
4.  * Copyright 2006-2010		Johannes Berg <johannes@sipsolutions.net>
5.  * Copyright 2013-2014  Intel Mobile Communications GmbH
6.  * Copyright 2015-2017	Intel Deutschland GmbH
7.  */
8. 
9. #define pr_fmt(fmt) KBUILD_MODNAME ": " fmt
10. 
11. #include <linux/if.h>
12. #include <linux/module.h>
13. #include <linux/err.h>
14. #include <linux/list.h>
15. #include <linux/slab.h>
16. #include <linux/nl80211.h>
17. #include <linux/debugfs.h>
18. #include <linux/notifier.h>
19. #include <linux/device.h>
20. #include <linux/etherdevice.h>
21. #include <linux/rtnetlink.h>
22. #include <linux/sched.h>
23. #include <net/genetlink.h>
24. #include <net/cfg80211.h>
25. #include "nl80211.h"
26. #include "core.h"
27. #include "sysfs.h"
28. #include "debugfs.h"
29. #include "wext-compat.h"
30. #include "rdev-ops.h"
31. 
32. /* name for sysfs, %d is appended */
33. #define PHY_NAME "phy"
34. 
35. MODULE_AUTHOR("Johannes Berg");
36. MODULE_LICENSE("GPL");
37. MODULE_DESCRIPTION("wireless configuration support");
38. MODULE_ALIAS_GENL_FAMILY(NL80211_GENL_NAME);
39. 
40. /* RCU-protected (and RTNL for writers) */
41. LIST_HEAD(cfg80211_rdev_list);
[... 1387 lines cut ...]
1429. out_fail_nl80211:
1430. 	unregister_netdevice_notifier(&cfg80211_netdev_notifier);
1431. out_fail_notifier:
1432. 	wiphy_sysfs_exit();
1433. out_fail_sysfs:
1434. 	unregister_pernet_device(&cfg80211_pernet_ops);
1435. out_fail_pernet:
1436. 	return err;
1437. }
1438. fs_initcall(cfg80211_init);
1439. 
1440. static void __exit cfg80211_exit(void)
1441. {
1442. 	debugfs_remove(ieee80211_debugfs_dir);
1443. 	nl80211_exit();
1444. 	unregister_netdevice_notifier(&cfg80211_netdev_notifier);
1445. 	wiphy_sysfs_exit();
1446. 	regulatory_exit();
1447. 	unregister_pernet_device(&cfg80211_pernet_ops);
1448. 	destroy_workqueue(cfg80211_wq);
1449. }
1450. module_exit(cfg80211_exit);

```
Thank you very much.

lwfinger, 2020-09-04:
I think I know at least part of the problem. This driver can create two wifi devices on a single radio. That is controlled by EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE in Makefile. On your system, it is set; however, your user-space code is only configuring one of the two, thus the warnings. I suspect that you want to comment out that line.

bigdolphin, 2020-09-05:
I commented that flag but the warning still occurs and module is not loaded.
When I enable the CONFIG_DEBUG, CONFIG_DEBUG_RTL871X and DBG_IOCTL, dmesg output says:

> [ 8686.173411] RTL871X: USB_SPEED_HIGH
> [ 8686.195628] RTL871X: CHIP TYPE: RTL8723BU
> [ 8686.199759] RTL871X: register rtw_netdev_ops to netdev_ops
> [ 8686.205425] RTL871X: rtw_wdev_alloc(padapter=c1601000)
> ..............
> [ 8686.215528] WARNING: CPU: 0 PID: 20205 at backports-4.19.137-1/net/wireless/core.c:743 wiphy_register+0x2a4/0x9d0 [cfg80211]
> ..............
> [ 8686.625526] WARNING: CPU: 0 PID: 20205 at backports-4.19.137-1/net/wireless/core.c:821 wiphy_register+0x61c/0x9d0 [cfg80211]
> ..............
> [ 8687.032036] RTL871X: Couldn't register wiphy device
> [ 8687.037041] RTL871X: rtw_usb_if1_init Failed!
> [ 8687.041646] usbcore: registered new interface driver rtl8723bu
> [ 8687.047602] RTL871X: module init ret=0

Best regards.

vbvchauthmal, 2022-04-28:
Hi @bigdolphin,
Did you able to resolve the issue?

I used same Makefiles and getting kernel panic on `modprobe rtl8723bu` module.
```
root@OpenWrt:/# lsusb 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@OpenWrt:/# 
root@OpenWrt:/# modprobe rtl8723bu
[ 1953.734110] RTL871X: module init start
[ 1953.738354] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[ 1953.745717] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[ 1953.890947] ------------[ cut here ]------------
[ 1953.896072] WARNING: CPU: 0 PID: 2222 at backports-4.19.221-1/net/wireless/reg.c:2339 wiphy_apply_custom_regulatory+0x3c/0x448 [cfg80211]
[ 1953.909272] wiphy should have REGULATORY_CUSTOM_REG
[ 1953.914420] Modules linked in: 8723bu(+) cfg80211 compat qcserial pppoe ppp_async option cdc_mbim usb_wwan qmi_wwan pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cde
[ 1953.990299]  usb_f_ecm u_ether libcomposite nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 uas usb_storage xhci_plat_hcd xhci_hcd ohci_platform ohci_m
[ 1954.011488] CPU: 0 PID: 2222 Comm: modprobe Not tainted 4.14.267 #0
[ 1954.018080] Hardware name: Generic AM33XX (Flattened Device Tree)
[ 1954.024487] Backtrace: 
[ 1954.027097] [<c010b598>] (dump_backtrace) from [<c010b890>] (show_stack+0x18/0x1c)
[ 1954.035061]  r7:bf349884 r6:60000013 r5:00000000 r4:c0d5ed5c
[ 1954.041029] [<c010b878>] (show_stack) from [<c0812a1c>] (dump_stack+0x9c/0xb0)
[ 1954.048641] [<c0812980>] (dump_stack) from [<c0131df4>] (__warn+0xec/0x108)
[ 1954.055965]  r7:bf349884 r6:bf375b88 r5:00000000 r4:de45db90
[ 1954.061924] [<c0131d08>] (__warn) from [<c0131e50>] (warn_slowpath_fmt+0x40/0x48)
[ 1954.069798]  r9:ddf3e470 r8:bf41d498 r7:bf415998 r6:de6f02e0 r5:de6f6000 r4:bf375c54
[ 1954.078077] [<c0131e14>] (warn_slowpath_fmt) from [<bf349884>] (wiphy_apply_custom_regulatory+0x3c/0x448 [cfg80211])
[ 1954.089136]  r3:00000000 r2:bf375c54
[ 1954.092900]  r4:de6f02e0
[ 1954.096234] [<bf349848>] (wiphy_apply_custom_regulatory [cfg80211]) from [<bf3db0a0>] (rtw_regd_init+0x30/0x3c [8723bu])
[ 1954.107693]  r10:de6f100c r9:ddf3e470 r8:bf41d498 r7:ddb41000 r6:e0603000 r5:de6f6000
[ 1954.115938]  r4:de6f02e0
[ 1954.119160] [<bf3db070>] (rtw_regd_init [8723bu]) from [<bf3dad84>] (rtw_wdev_alloc+0x1b0/0x2d8 [8723bu])
[ 1954.129237]  r5:de6f6000 r4:de6f02e0
[ 1954.133500] [<bf3dabd4>] (rtw_wdev_alloc [8723bu]) from [<bf3c92e4>] (rtw_drv_init+0x38c/0x4cc [8723bu])
[ 1954.143492]  r9:ddf3e470 r8:ddb41000 r7:e0603000 r6:ddf3e800 r5:e0603000 r4:de6f1000
[ 1954.151915] [<bf3c8f58>] (rtw_drv_init [8723bu]) from [<c05e7a5c>] (usb_probe_interface+0x1b8/0x258)
[ 1954.161524]  r10:bf41d2b8 r9:ddf3e800 r8:00000000 r7:ddf3e400 r6:bf41d268 r5:ddf3e470
[ 1954.169755]  r4:ddf3e820
[ 1954.172444] [<c05e78a4>] (usb_probe_interface) from [<c05126f8>] (driver_probe_device+0x158/0x2f0)
[ 1954.181869]  r10:00000000 r9:00000010 r8:bf41d268 r7:00000000 r6:c12892cc r5:c12892c8
[ 1954.190099]  r4:ddf3e820
[ 1954.192778] [<c05125a0>] (driver_probe_device) from [<c0512920>] (__driver_attach+0x90/0xb4)
[ 1954.201655]  r10:c0d07548 r9:00000000 r8:bf438bc0 r7:00000000 r6:ddf3e854 r5:bf41d268
[ 1954.209886]  r4:ddf3e820 r3:00000000
[ 1954.213658] [<c0512890>] (__driver_attach) from [<c0510af4>] (bus_for_each_dev+0x54/0xa8)
[ 1954.222258]  r7:00000000 r6:c0512890 r5:bf41d268 r4:00000000
[ 1954.228217] [<c0510aa0>] (bus_for_each_dev) from [<c0512150>] (driver_attach+0x24/0x28)
[ 1954.236634]  r6:c0d4800c r5:ddb57900 r4:bf41d268
[ 1954.241500] [<c051212c>] (driver_attach) from [<c0511bf0>] (bus_add_driver+0xf0/0x208)
[ 1954.249831] [<c0511b00>] (bus_add_driver) from [<c0513078>] (driver_register+0xb0/0xec)
[ 1954.258249]  r7:bf419b80 r6:bf41d268 r5:c0d4800c r4:bf41d268
[ 1954.264209] [<c0512fc8>] (driver_register) from [<c05e6480>] (usb_register_driver+0x8c/0x144)
[ 1954.273171]  r5:c0d4800c r4:bf41d218
[ 1954.277184] [<c05e63f4>] (usb_register_driver) from [<bf45a0a0>] (init_module+0xa0/0x1000 [8723bu])
[ 1954.286701]  r9:00000000 r8:bf45a000 r7:dd5976c0 r6:ffffe000 r5:00000000 r4:bf41d218
[ 1954.295080] [<bf45a000>] (init_module [8723bu]) from [<c0101b30>] (do_one_initcall+0xc8/0x18c)
[ 1954.304154]  r5:00000000 r4:c0d07548
[ 1954.307945] [<c0101a68>] (do_one_initcall) from [<c01abd80>] (do_init_module+0x68/0x208)
[ 1954.316475]  r9:00000000 r8:bf438bc0 r7:dd5976c0 r6:ddc94258 r5:ddc94000 r4:bf438bc0
[ 1954.324639] [<c01abd18>] (do_init_module) from [<c01aaef8>] (load_module+0x1ea4/0x2280)
[ 1954.333074]  r7:ddc94288 r6:ddc94258 r5:ddc94000 r4:de45df30
[ 1954.339046] [<c01a9054>] (load_module) from [<c01ab418>] (SyS_init_module+0x144/0x17c)
[ 1954.347392]  r10:00000051 r9:000128de r8:ffffe000 r7:e0601574 r6:b6e60584 r5:00000000
[ 1954.355637]  r4:0000c574
[ 1954.358329] [<c01ab2d4>] (SyS_init_module) from [<c0107c60>] (ret_fast_syscall+0x0/0x54)
[... 232 lines cut ...]
[ 1956.384228] [<c0512fc8>] (driver_register) from [<c05e6480>] (usb_register_driver+0x8c/0x144)
[ 1956.393199]  r5:c0d4800c r4:bf41d218
[ 1956.397200] [<c05e63f4>] (usb_register_driver) from [<bf45a0a0>] (init_module+0xa0/0x1000 [8723bu])
[ 1956.406712]  r9:00000000 r8:bf45a000 r7:dd5976c0 r6:ffffe000 r5:00000000 r4:bf41d218
[... cut ...]
[ 1956.486773]  r10:00000080 r9:de45c000 r8:c0107e64 r7:00000080 r6:00000003 r5:00000000
[ 1956.495012]  r4:00000000
[ 1956.497691] Code: 1a000003 e5943014 e3130001 1a000000 (e7f001f2) 
[ 1956.504115] ---[ end trace 4f62d3a66bbf2884 ]---
[ 1956.508979] Kernel panic - not syncing: Fatal exception
[ 1956.514502] Rebooting in 3 seconds..
```

I am compiling this driver for Beaglebone Black based platform with OpenWrt version [v19.07.9](https://github.com/openwrt/openwrt/tree/v19.07.9).

@lwfinger I have commented `EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE` in my compilation.

bigdolphin, 2022-04-28:
> Hi @bigdolphin, Did you able to resolve the issue?
>
> I used same Makefiles and getting kernel panic on `modprobe rtl8723bu` module.
>
> ```
> root@OpenWrt:/# lsusb
> Bus 002 Device 001: ID 1d6b:0002 Linux [...]

Hi, I gave up using rtl8723bu because the driver source is not compatible with new gcc of OpenWrt v19 (as I remember). I changed to use rtl8723du which is compatible with gcc 7.5.0. RTL8723du is totally same with bu version but its driver source is newer. I confirm that OpenWRT v19.07 works well with rtl8723du.

lwfinger, 2022-04-28:
@vbvchauthmal - It appears as if you are using kernel 4.19. Is that correct? Which branch of the repo? As far as I can see, the first warning about not having REGULATORY_CUSTOM_REG set is bogus as the code definitely sets it. I have not looked into the rest of the warnings or the oops yet.

lwfinger, 2022-05-01:
What branch of this repo are you using? The 'git branch' command will tell you that information.

vbvchauthmal, 2022-05-02:
@lwfinger I am using the `master` branch of this repo having last commit id `2cd9874d128423c92314324abc7ee0fd7923afcf`.