lwfinger/rtl8723bu #170: Kernel 4.1.15 Can only compile the driver as a module but not into the kernel?

TelDragon, 2020-09-29:
Kernel 4.1.15 I loaded the driver as a module and it was perfect. I tried to select "*" to write the kernel and compile it together, but it was not recognized after being burned into the board.

lwfinger, 2020-09-29:
If you want the driver to be built into the kernel, you would need to merge the code with kernel source, probably in the drivers/staging tree. All statements involving #LINUX_VERSION_CODE would need to be simplified so that only the code appropriate for your kernel is left. Finally, you would need to write a Kconfig and Makefile suitable for kernel building. It would take an expert about 10 hours to complete that. I suggest you find out how to add an external module to your system. That would be a lot easier.

TelDragon, 2020-09-29:
Thank you very much for your attention.

I will automatically load this driver according to modules

* Edit loading file

```Shell
vi /etc/modules

8723bu
```

* Add kernel file boot

```Shell
mkdir -p /lib/modules/$(uname -r)/kernel/drivers/net/
cp 8723bu.ko /lib/modules/$(uname -r)/kernel/drivers/net/
```

* Generate module load file

```Shell
cp kernel/modules.builtin /lib/modules/$(uname -r)/kernel
cp kernel/modules.order /lib/modules/$(uname -r)/kernel
depmod -a
```

TelDragon, 2020-09-29:
But I have another problem

The drive speed provided by the original factory can reach nearly 10M/s.

http://mirrors.163.com/debian-cd/10.6.0/armhf/iso-dvd/debian-10.6.0-armhf-DVD-1.iso

Test

Using your drive is much slower 2M/s

lwfinger, 2020-09-29:
That I cannot help you with. I did not write the driver - I only provide the factory driver here in a form that will build. I do not understand the details of the chip! Sorry.

closed 2020-09-29 by lwfinger