lwfinger/rtl8723bu #174: Turning off and on the card stop to connect to the AP

vito-r1, 2021-02-12:
Hello,
I have just built this driver and I found an issue that is worrying me.
After turning off and after that turning it on trough Network Manager the driver stop to connect to the AP so I am pasting you the information that is from dmesg and regarding the system.
Thank you for your time to check this issue.

5.10.0-3-amd64 #1 SMP Debian 5.10.13-1 (2021-02-06) x86_64 GNU/Linux

[   18.548564] RTL871X: module init start
[   18.548569] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   18.548570] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   18.674924] RTL871X: rtw_ndev_init(wlan0)
[   18.675226] usbcore: registered new interface driver rtl8723bu
[   18.675228] RTL871X: module init ret=0
[   18.696551] Bluetooth: hci0: RTL: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[   18.697458] Bluetooth: hci0: RTL: rom_version status=0 version=1
[   18.697460] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723b_fw.bin
[   18.698382] bluetooth hci0: firmware: direct-loading firmware rtl_bt/rtl8723b_fw.bin
[   18.698390] rtl8723bu 1-4:1.2 wlx347563255c4b: renamed from wlan0
[   18.698408] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723b_config.bin
[   18.698423] bluetooth hci0: firmware: failed to load rtl_bt/rtl8723b_config.bin (-2)
[   18.698524] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[   18.698535] Bluetooth: hci0: RTL: cfg_sz -2, total sz 22496
[   19.139463] Bluetooth: hci0: RTL: fw version 0x0e2f9f73

After turning it off and on

[ 5453.989801] RTL871X: nolinked power save leave
[ 5456.081848] RTL871X: nolinked power save enter
[ 5473.620429] RTL871X: RTW_ADAPTIVITY_EN_
[ 5473.620434] AUTO, chplan:0x20, Regulation:0,0
[ 5473.620435] RTL871X: RTW_ADAPTIVITY_MODE_
[ 5473.620435] NORMAL
[ 5473.994126] RTL871X: nolinked power save leave
[ 5476.093307] RTL871X: nolinked power save enter
[ 5502.640148] RTL871X: RTW_ADAPTIVITY_EN_
[ 5502.640153] AUTO, chplan:0x20, Regulation:0,0
[ 5502.640154] RTL871X: RTW_ADAPTIVITY_MODE_
[ 5502.640154] NORMAL
[ 5503.014248] RTL871X: nolinked power save leave
[ 5505.104642] RTL871X: nolinked power save enter

lwfinger, 2021-02-12:
There is no attempt to connect shown in the second messages. It is just going in and out of power save. I have not heard of any power-save problems, but you can turn it off with the module option "rtw_power_mgnt=0'.

vito-r1, 2021-02-13:
Hello,
I tried with this option but there is no change.

options 8723bu rtw_power_mgnt=0

[  332.037456] RTL871X: RTW_ADAPTIVITY_MODE_
[  332.037456] NORMAL
[  332.411923] RTL871X: nolinked power save leave
[  334.507348] RTL871X: nolinked power save enter
[  361.060414] RTL871X: RTW_ADAPTIVITY_EN_
[  361.060418] AUTO, chplan:0x20, Regulation:0,0
[  361.060419] RTL871X: RTW_ADAPTIVITY_MODE_
[  361.060420] NORMAL
[  361.434007] RTL871X: nolinked power save leave
[  363.524747] RTL871X: nolinked power save enter

vito-r1, 2021-02-22:
Hello, I must update the name of the issue - it happens when the AP is disconnected not only when the card is turned off.
You can see the log that I retrieved (only the MAC address is removed). If you know where can I find more info at the logs please tell me. My workaround has now been just to remove the driver and loaded it again and the connection happens.

 Feb 22 09:14:19  kernel: RTL871X: nolinked power save enter
Feb 22 09:14:26  kernel: RTL871X: nolinked power save leave
Feb 22 09:14:26  NetworkManager[674]: <info>  [1613981666.5766] device (wlx347563255c4b): supplicant interface state: disconnected -> scanning
Feb 22 09:14:26  NetworkManager[674]: <info>  [1613981666.5767] device (p2p-dev-wlx347563255c4b): supplicant management interface state: disconnected -> scanning
Feb 22 09:14:28  wpa_supplicant[690]: wlx347563255c4b: CTRL-EVENT-SSID-REENABLED id=0 ssid="P"
Feb 22 09:14:28  wpa_supplicant[690]: wlx347563255c4b: Trying to associate with xx:xx (SSID='P' freq=2462 MHz)
Feb 22 09:14:28  wpa_supplicant[690]: wlx347563255c4b: Association request to the driver failed
Feb 22 09:14:28  wpa_supplicant[690]: wlx347563255c4b: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="P" auth_failures=2 duration=20 reason=CONN_FAILED
Feb 22 09:14:28  NetworkManager[674]: <info>  [1613981668.2690] device (wlx347563255c4b): supplicant interface state: scanning -> disconnected
Feb 22 09:14:28  NetworkManager[674]: <info>  [1613981668.2691] device (p2p-dev-wlx347563255c4b): supplicant management interface state: scanning -> disconnected
Feb 22 09:14:32  kernel: RTL871X: nolinked power save enter
Feb 22 09:14:39  kernel: RTL871X: nolinked power save leave

ivnish, 2021-04-06:
Same problem. After sleep I can't connect to AP. Only reboot helps

ivnish, 2021-04-06:
https://github.com/lwfinger/rtl8723bu/issues/152#issuecomment-708210789

lwfinger, 2021-04-06:
Does 'sudo modprobe -r 8723bu' before sleep, followed by 'sudo modprobe 8723bu' after sleep restore network? If so, then the service script in #152 will work.

mikaku, 2022-04-26:
Hi,

I've a Raspberry Pi 1 Model A+ with a dongle USB Bluetooth/WiFi using the driver rtl8xxxu which in turn automatically installs the driver rtl8723bu and I have the same problem. It does not reconnect after the first disconnection.

The work-around to remove and reinstall the driver rtl8xxxu is not a valid option here because it also affects the Bluetooth driver, since this is a dongle with Bluetooth and WiFi all managed with the driver rtl8xxxu.

How can I fix this problem?

lwfinger, 2022-04-26:
If you are using rtl8xxxu to control your wifi, then you are asking in the wrong place. You should be asking on the linux-wireless mailing list at linux-wireless@vger.kernel.org. Note: That ML will not accept html, only plain text.