lwfinger/rtl8723bu #179: 4.3.16 branch has no dkms.conf

kevin201, 2021-08-06:
"Note: The code in branch v4.3.16 is better than that in master. Get that with
git clone https://github.com/lwfinger/rtl8723bu.git -b v4.3.16"

Is this still valid? The 4.3.16 branch has no dkms.conf 

I'm trying to solve a problem with the device not being recognised after the kernel upgrade to 5.11.0-25

dkms has been working perfectly on every previous kernel upgrade. (5.4, 5,8)

dkms status
rtl8723bu, 4.3.6.11_12942.20141204_BTCOEX20140507-4E40, 5.4.0-80-generic, x86_64: installed
rtl8723bu, 4.3.6.11_12942.20141204_BTCOEX20140507-4E40, 5.8.0-63-generic, x86_64: installed

lwfinger, 2021-08-06:
I do not do dkms. The only way that dkms gets into a repo is because someone else sent me the necessary patches.

I pushed the fixes for later kernels, but that particular release still does not build here. Please use the master branch.

closed 2021-08-06 by lwfinger

kevin201, 2021-08-06:
I thought I was using the master branch. It fails regardless of dmks.