lwfinger/rtl8723bu #182: help to build inside android kernel tree

Duhjoker, 2021-09-08:
hi 

 working on an Android kernel msm-4.9 for pixel 3a and need to add this inside kernel tree as theres no other way to build it correctly. could you please provide instruction on how to do so. thank you

lwfinger, 2021-09-09:
I do not understand the question. This driver should build on a 4.9 kernel.

Biohazardousrom, 2023-09-21:
sorry i have since upgraded my phone and now im working with a linux-5.10 kernel aarch64. i have placed this driver in /drivers/staging with the rest of the rtl drivers i have, adjusted the makefile and kconfig as so and corrected a cflag so that it sees its own includes and the top of the tree /includes 

i can get it to vmlinux at lto but at modpost i get this error

[code]
ERROR: modpost: module 8723bu uses symbol filp_open from namespace VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
make[2]: *** [/home/duhjoker/pantah-kernel-ex/private/gs-google/scripts/Makefile.modpost:152: modules-only.symvers] Error 1
make[2]: *** Deleting file 'modules-only.symvers'
make[1]: *** [/home/duhjoker/pantah-kernel-ex/private/gs-google/Makefile:1501: modules] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [/home/duhjoker/pantah-kernel-ex/private/gs-google/Makefile:185: __sub-make] Error 2

[code/]

can you please advise me how to continue

please see my commit history for what exactly i have done

https://github.com/Biohazardousrom/Mad-Kali-MaxHunter-Pantah/tree/QPR2-release-test-drivers-rtl

lwfinger, 2023-09-22:
Sorry it took a while to answer your problem. For Android, an MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
macro call is needed. As I said in the commit message, it does not seem to hurt Linux builds.