lwfinger/rtl8723bu #186: make install errors - Ubuntu 21 - kernel 5.13.0-27-generic - Intel(R) Celeron(R) N4020 CPU @ 1.10GHz

NVG2021, 2022-02-02:
Hello,
I've been trying to install the rtl8723bu drivers on my laptop with linux Ubuntu 21 and I get the following messages.
during the make process there are two or three misleading statements and for the make install errors
Would you give any hint on how to avoid the error???

duser@prixton:~/rtl8723bu$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-27-generic/build M=/home/aduser/rtl8723bu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-27-generic'
  CC [M]  /home/aduser/rtl8723bu/core/rtw_cmd.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_security.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_debug.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_io.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_ioctl_query.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_ioctl_set.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_ieee80211.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_mlme.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_mlme_ext.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_wlan_util.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_vht.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_pwrctrl.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_rf.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_recv.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_sta_mgt.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_ap.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_xmit.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_p2p.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_tdls.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_br_ext.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_iol.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_sreset.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_btcoex.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_beamforming.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_odm.o
  CC [M]  /home/aduser/rtl8723bu/core/rtw_efuse.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/osdep_service.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/os_intfs.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/usb_intf.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/usb_ops_linux.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/ioctl_linux.o
/home/aduser/rtl8723bu/os_dep/ioctl_linux.c: In function ‘rtw_wx_set_enc’:
/home/aduser/rtl8723bu/os_dep/ioctl_linux.c:2816:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
 2816 |                 if(rf_on == pwrpriv->rf_pwrstate )
      |                 ^~
/home/aduser/rtl8723bu/os_dep/ioctl_linux.c:2818:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
 2818 |                         ret = -EOPNOTSUPP;
      |                         ^~~
  CC [M]  /home/aduser/rtl8723bu/os_dep/xmit_linux.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/mlme_linux.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/recv_linux.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/ioctl_cfg80211.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/wifi_regd.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/rtw_android.o
  CC [M]  /home/aduser/rtl8723bu/os_dep/rtw_proc.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_intf.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_com.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_com_phycfg.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_phy.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_btcoex.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_usb.o
  CC [M]  /home/aduser/rtl8723bu/hal/hal_usb_led.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalPwrSeqCmd.o
  CC [M]  /home/aduser/rtl8723bu/hal/Hal8723BPwrSeq.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_sreset.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_hal_init.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_phycfg.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_rf6052.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_dm.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_rxdesc.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723b_cmd.o
  CC [M]  /home/aduser/rtl8723bu/hal/usb_halinit.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723bu_led.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723bu_xmit.o
  CC [M]  /home/aduser/rtl8723bu/hal/rtl8723bu_recv.o
  CC [M]  /home/aduser/rtl8723bu/hal/usb_ops.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_debug.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_AntDiv.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_interface.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_HWConfig.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalPhyRf.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_EdcaTurboCheck.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_DIG.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_PathDiv.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_RaInfo.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_DynamicBBPowerSaving.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_DynamicTxPower.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_CfoTracking.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_NoiseMonitor.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalBtc8723b1Ant.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalBtc8723b2Ant.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalHWImg8723B_BB.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalHWImg8723B_MAC.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalHWImg8723B_RF.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalHWImg8723B_FW.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_RegConfig8723B.o
  CC [M]  /home/aduser/rtl8723bu/hal/HalPhyRf_8723B.o
  CC [M]  /home/aduser/rtl8723bu/hal/odm_RTL8723B.o
  CC [M]  /home/aduser/rtl8723bu/platform/platform_ops.o
  LD [M]  /home/aduser/rtl8723bu/8723bu.o
  MODPOST /home/aduser/rtl8723bu/Module.symvers
  CC [M]  /home/aduser/rtl8723bu/8723bu.mod.o
  LD [M]  /home/aduser/rtl8723bu/8723bu.ko
  BTF [M] /home/aduser/rtl8723bu/8723bu.ko
Skipping BTF generation for /home/aduser/rtl8723bu/8723bu.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-27-generic'
aduser@prixton:~/rtl8723bu$ sudo make install
[sudo] password for aduser: 
install -p -m 644 -D 8723bu.ko /lib/modules/5.13.0-27-generic/kernel/drivers/net/wireless/8723bu.ko
/sbin/depmod -a 5.13.0-27-generic
depmod: ERROR: failed to load symbols from /lib/modules/5.13.0-27-generic/kernel/drivers/regulator/lp8755.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.13.0-27-generic/kernel/drivers/comedi/drivers/dt2817.ko: Invalid argument
install rtl8723b_fw.bin -D /lib/firmware/rtl_bt/rtl8723b_fw.bin

lwfinger, 2022-02-02:
You do not need to worry about those errors. Two other drivers in your system, lp8755 and dt2817, are messed up, but they will not affect your wireless.

The warning during the build can also be ignored.

NVG2021, 2022-02-02:
thanks for the reply... and now I can follow the readme.. no?
many thanks

lwfinger, 2022-02-02:
I thought you were following the readme.

NVG2021, 2022-02-02:
yes, i did, just stopped with the errors, will proceed
so, when last input for blacklist is done, reboot?

NVG2021, 2022-02-02:
I've rebooted but the wifi does not show on the network options..... one thing I've noticed is that my device number does not match with what the readme.md file says...
Mines is Bus 001 Device 003: ID 0bda:d723 for a Realtek and on the readme it says `0bda:b720` almost the same, but different number at the end...
any hint? Many thanks again

lwfinger, 2022-02-03:
The USB ID differs only by 3 at the end, but 0bda:d723 is for an rtl8723du, not an rtl8723bu. You have installed the wrong driver. Use git clone git://GitHub.com/lwfinger/rtl8723du.git -b v5.13.4 to get the correct one.

I am closing this issue. If you still have problems, open a new issue in the rtl8723du repo.

closed 2022-02-03 by lwfinger