lwfinger/rtl8723bu #188: Driver error installing in kernel 5.13.0, ubuntu 21.10

fazaldegui, 2022-04-09:
Hi,
I follow the steps in Readme.md. During make a warning appear:
` CC [M]  /home/holst/Programas/USB_Edimax_WiFi_Bluetooth/driver_repo_git/rtl8723bu/8723bu.mod.o
  LD [M]  /home/holst/Programas/USB_Edimax_WiFi_Bluetooth/driver_repo_git/rtl8723bu/8723bu.ko
  BTF [M] /home/holst/Programas/USB_Edimax_WiFi_Bluetooth/driver_repo_git/rtl8723bu/8723bu.ko
Skipping BTF generation for /home/holst/Programas/USB_Edimax_WiFi_Bluetooth/driver_repo_git/rtl8723bu/8723bu.ko due to unavailability of vmlinux
make[1]: se sale del directorio '/usr/src/linux-headers-5.13.0-39-generic'
ho`

In the make install no error no warnings.

But fails at modprobe:
$ sudo modprobe -v 8723bu
insmod /lib/modules/5.13.0-39-generic/kernel/drivers/net/wireless/8723bu.ko 
modprobe: ERROR: could not insert '8723bu': Invalid argument

I've also tried the "Automatic install using DKMS" but no luck, I don't see the wifi adapter.

Any idea?
Thank you

lwfinger, 2022-04-09:
That warning is harmless. You should always look in the dmesg output to see the actual error.

Are you using secure boot? If so, do a 'git pull' and do 'sudo make sign-install'.

fazaldegui, 2022-04-10:
First of all, thank you your answer and help! :)

I'm not using Secure Boot, in the BIOS shows Disabled.
But I did what you said: 
* git pull: I'm already update
* sudo make sign-install: No rules with that name
The error with the modprobe continues:
$ sudo modprobe -v 8723bu
insmod /lib/modules/5.13.0-39-generic/kernel/drivers/net/wireless/8723bu.ko 
modprobe: ERROR: could not insert '8723bu': Invalid argument

[Link to dmesg output](https://pastebin.com/8Eu4J05p)

BTW, I tried the driver in Edimax download page:
https://www.edimax.com/edimax/download/download/data/edimax/es/download/undefined/for_home/wireless_adapters_n150/ew-7611ulb
And with the bluetooth no problems, I've installed it correctly but the WiFi gives error at "make" step: errors at compile time:
[Link to compile error of Edimax driver](https://pastebin.com/Z9NjDmif)

Thank you

lwfinger, 2022-04-10:
First of all, I am not going to fix the Edimax drivers. This repo started from the same Realtek driver and has many changes that I would have to repeat for no real gain!

In the future, if a dmesg output is required, use 'dmesg > dmesg.txt' and attach dmesg.txt to the issue.
Your dmesg output shows many lines of the form
"[ 2552.006801] 8723bu: disagrees about version of symbol cfg80211_rx_mgmt_khz"

That means that the kernel headers you used do not match the running kernel. You can see that value with 'uname -r'. That MUST MATCH EXACTLY with the output on the make output that starts with "make[1]: Entering directory..."

fazaldegui, 2022-04-10:
Hi,
I'm sorry if I've bothered you in any way. I only show that if it could help, not to fix an external software. Sorry for the misunderstanding.
To attach the trace file to the issue I agree. Before attach I see in other issues that a way of attach the traces was using pastebin.com and I did the same. Of course, before put this issue I've used the search to tried find something similar...

So, the main subject here is to give thank you to your work, and that's I've tried to do before and after talking about my issues. I've started this issue only to tried to be constructive, because my first idea was to throw this hardware to my digital gadgets drawer of memories.

So, to end this issue I've attach the make trace to see that the kernel version MATCH EXACTLY with the output of the command 'uname -r':
$ uname -r
5.13.0-39-generic
[make.txt](https://github.com/lwfinger/rtl8723bu/files/8459649/make.txt)

Now, I'll go back to my first idea and buy a hardware that could work on linux without this several issues.
Thank you for your time.

lwfinger, 2022-04-10:
This is your choice.

closed 2022-04-10 by lwfinger