lwfinger/rtl8723bu #190: compile error with linux519: incompatible-pointer-types

oberon-manjaro, 2022-08-18:
```
src/rtl8723bu-ff22a9fc36595db4c66b7bdc3f7646d70a978a45/os_dep/ioctl_cfg80211.c:5905:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
 5905 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/bernhard/manjaro/extra/linux519-extramodules/rtl8723bu/src/rtl8723bu-ff22a9fc36595db4c66b7bdc3f7646d70a978a45/os_dep/ioctl_cfg80211.c:5905:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)

```
full build log attached
[rtl8723bu-202208171911.log](https://github.com/lwfinger/rtl8723bu/files/9373926/rtl8723bu-202208171911.log)

oberon-manjaro, 2022-08-18:
This issue starts with kernel 5.19.2 - compiling with 5.19.1 was still fine

dini, 2022-08-18:
fix in #191