lwfinger/rtl8723bu #197: Not working in kernel 6.1.9

eyrent, 2023-02-06:
- Fedora 37 had an update after which wifi stopped working.
- Blacklisted 8723bu and used rtl8xxxu. NetworkManages seemed to make use of the adaptor but speed problems for which I had turned to this repo driver remained, as I imagined.
- Carried on instructions to install using the latest version of master here but wifi never returned. Not even after restoring rtl8xxxu. So muy laptop is connectionless and I don't know how to fix this. 
- lsusb and inxi show the adaptor
- rfkill shows a soft kill (probably due to driver not working)

eyrent, 2023-02-06:
Update: I could revert back to rtl8xxxu and got the adaptor in networkmanager again.

lwfinger, 2023-02-06:
I rechecked the code, and could not find any problem with newer kernels. Run this driver, do the command
sudo dmesg -t > dmesg.txt
and attach dmesg.txt.

eyrent, 2023-02-06:
I'm a bit awkward to admit it but driver is working fine. I don't know it this was all caused by dkms (which I uninstalled after removing dkms config for this driver). I apologize for taking you time.

lwfinger, 2023-02-06:
Glad to hear that. I do not trust dkms at all.

closed 2023-02-06 by lwfinger