lwfinger/rtl8723bu #206: Take Over

giuliobenetti, 2025-07-01:
This is to let people know that my Company has taken over this Repository:
https://github.com/Benetti-Engineering-sas/rtl8723bu

And we will keep maintaining it with latest Linux Kernels until the mainline driver will be completed.

RIP Larry Finger

SubSpaceManeuvers, 2025-09-08:
Thankful for your continuation of the work. Build is failing under 6.16, is there a fix in the works perchance? I am sorry for asking here but issues are apparently disabled on your fork of the code.

giuliobenetti, 2025-09-08:
> Thankful for your continuation of the work. Build is failing under 6.16, is there a fix in the works perchance? I am sorry for asking here but issues are apparently disabled on your fork of the code.

Hi @SubSpaceManeuvers, thank you for pointing me that that Repository has no Issue enabled. Some time ago Github enabled it by default. I ask kindly you to open an Issue here:
https://github.com/Benetti-Engineering/rtl8723bu/issues

Thank you!